bambulab/BambuStudio #3203: Flying bottom support interface

man55, 2023-12-22:
### Bambu Studio Version

1.8.2

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C combo

### How to reproduce

Use Bambu Support for PLA (formerly 'Support W') Filament for intarface, which starts from the printing part.
[image]
[image]

### Actual results

Before printing the support interface, the Bambu Studio slicer skips one layer. As a result, the support interface does not fit onto the part, but literally disintegrates into threads. Subsequently, this support cannot be held on the part and is easily broken off by the nozzle, the part is damaged.
Support settings in Bambu Studio do not solve the problem.

I have researched previous beta releases and can say that the bug was introduced starting with version 1.8.0 Public beta

### Expected results

This is how it worked before in version 1.7.7.
The support interface must be applied directly to the part without skipping layers. Then such support is sustainable.
[image]

### Project file & Debug log uploads

[982398437898239.zip](https://github.com/bambulab/BambuStudio/files/13756992/982398437898239.zip)
This is a trimmed part for display purposes.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

man55, 2023-12-23:
Did some extensive research.
I created a simple model for clarity.
Now step by step, starting with the standard profile of Bamboo Studio version 1.8.2.

[image]
stl [Test bottom support interface v2.zip](https://github.com/bambulab/BambuStudio/files/13759124/Test.bottom.support.interface.v2.zip)

1. Just enable support, default settings. OK here.
[image]

2. Set Bottom interface spacing to 0. Slice. Nothing changed.
[image]

3. Set Bottom Z distance to 0. Slice. Nothing changed.
[image]

4. Set Top interface spacing to 0. Slice. Nothing changed.
[image]

5. Set Top Z distance to 0. Slice. Finally, the gap at the top disappeared. But what to do with the gap at the bottom? How to remove it?
[image]

6. Set Filament for Support Interface to Bambu Support PLA, which is the goal here. This material has low adhesion for easy release after printing. And therefore it should be printed without gaps.
But the gap is still there and there is no way to remove it with the settings.
[image]

man55, 2023-12-23:
It turns out that you can set a negative value for the "Top Z distance", although this seems illogical. And if the modulus of the number is equal to the height of the layer, then a miracle happens and the gap at the bottom disappears.
That is, by changing the value for the TOP, the effect is felt at the BOTTOM.
[image]

But then another problem arises - the support sticks together into the model. 
[image]

The interface lines literally overlap the lines of the model, and the support base close to the model without a gap.
And the horizontal gap parameter "Support/object xy distance" stops working altogether. Any value does not change anything.
[image]
[image]

tangketan, 2023-12-25:
Yes, it's a bug. We'll fix it soon.