bambulab/BambuStudio #3204: When printing PETG with multiple colors, the nozzle temperature is set to 255 degrees Celsius. However, during filament purging and material change, the temperature is lowered to 220 degrees Celsius.

jdnt, 2023-12-23:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

The PETG consumables use the same parameters as Bambu PETG Basic. The model can be found at: https://makerworld.com/zh/models/58861#profileId-60690. Please use the author's settings and replace the material with PETG. Set the purge factor to 0.65 to reduce the number of purges.The issue occurs during the material change process. While printing PETG, the temperature is set to 255 degrees, but during the purging phase, it drops to 220 degrees.

### Actual results

The excessively low purging temperature, when the cooling fan is activated, can cause premature cooling and hardening of the waste filament, resulting in a larger occupied space volume and potential blockage of the waste discharge outlet.

### Expected results

During multi-color printing and filament change, the purging temperature should be at least the same as the printing temperature, rather than lower.

### Project file & Debug log uploads

The second and third images show the filament purging process during a material change, where it can be observed that the system starts purging after lowering the temperature from 255 to 220 degrees. The fourth image represents the temperature during regular printing.
[image]
[image]
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jdnt, 2023-12-23:
[雪人灯.gcode.zip](https://github.com/bambulab/BambuStudio/files/13758422/gcode.zip)

djngnear, 2023-12-23:
I had been wondering about this as well. I hope it gets addressed.

jdnt, 2023-12-25:
The issue has been resolved, and the service ticket for the China region has provided an explanation.

[image]

"The reason for modifying the purge temperature is due to the higher viscosity of PETG. If purged at high temperatures, the ejected waste may stick together and get stuck in the waste bin. Therefore, unlike other materials that use maximum printing temperatures for purging, after testing, a specific purge temperature of 220 degrees Celsius has been set solely for PETG. Feel free to delete or modify the g-code below for adjustments if needed."

closed 2023-12-25 by jdnt