bambulab/BambuStudio #3205: Not possible to add diffrent Filament-Material the the listet like ASA, ABS,PLA, TPU, PVA, PA or PC

LucynaKushinada42, 2023-12-23:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Printer

P1S

### How to reproduce

Try to add custom-Material-Typ.
[image]
Insert Manufacturer, try to typ in custom Filament-Material-Typ.

### Actual results

Not possible to add for example BVOH. 

### Expected results

Add BVOH as custom Material!

### Project file & Debug log uploads

Not possible

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-12-25:
#3205

SaltWei, 2024-01-18:
Thanks. We will add into list

XunZhangBambu, 2024-06-26:
[image]
we have add many other filament types，so i think this issue can be closed now

closed 2024-06-26 by XunZhangBambu