bambulab/BambuStudio #3208: strange preview

progl, 2023-12-23:
### Bambu Studio Version

1 8 2 56

### Where is the application from?

Bambu Lab Official website

### OS version

macOS

### Additional system information

M1 Max

### Printer

bambu carbon

### How to reproduce

import str and make preview -

### Actual results

 no first layer

### Expected results

have first layer

### Project file & Debug log uploads

[debug_Sat_Dec_23_13_56_57_89998.log.0.zip](https://github.com/bambulab/BambuStudio/files/13758963/debug_Sat_Dec_23_13_56_57_89998.log.0.zip)
[mesh_0250um.3mf.zip](https://github.com/bambulab/BambuStudio/files/13758964/mesh_0250um.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

progl, 2023-12-23:
[image]
[image]

djngnear, 2023-12-23:
The mesh might be too thin for the nozzle. Have you tried scaling the model up, using arachne or setting your nozzle to 0.2mm?

QingZhangBambu, 2023-12-25:
@progl exactly as @djngnear said.