bambulab/BambuStudio #3209: Editing and saving User Preset causes hang

xVolta, 2023-12-23:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

N/A

### Printer

X1C

### How to reproduce

Calibrating filament and/or saving a user preset fails when starting from a user preset.

For example, with a roll of filament that's underextruding with default settings, first I selected "Generic PLA" adjusted the settings to suit this filament, then saved as a user preset.  No problem yet.  I then ran through the Flow Dynamics calibration, got and saved my new K value, still no problem.  I then continued with the flow rate calibration, which succeeded, but when I tried to save the new user preset, Bambu studio hangs.  Made a note of the calculated flow ratio, crashed the app, and attempted to manually edit the flow ratio on my user preset.  Same problem, app hangs on save.

Workaround, I went back to "Generic PLA", adjusted all settings to match the previous user preset, but this time with the new flow rate.  Save out, and Robert's your mother's brother.

### Actual results

App Crashes

### Expected results

User filament preset created.

### Project file & Debug log uploads

Blank project, just adjusting user presets.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DanBao-Bambu, 2024-01-22:
Have you changed the name of the preset you saved during Flow Dynamics calibration? We have fixed an application hanging issue caused by saving a user preset with a different name in the latest Studio. You can try this step again. If this problem occurs again, please upload the user log. It is best to export the dmp file for us to analyze when hanging. The following are the export steps:
1. Open Task Manager
2. Select the suspended Studio
3. Right click and select Create Dump File
Send it to us together.
If the dump file is too large, you can use a cloud drive to share it with us. Thank you.

DanBao-Bambu, 2024-02-26:
Will the current problem still arise?