bambulab/BambuStudio #3210: MicroSD not accesible in LAN-only mode

zlice, 2023-12-23:
Using BambuStudio on Linux (1.8.2) says the microSD can't be accessed in LAN-only mode. I imagine it's just a file protocol over wifi and easy enough to view files. I was trying to delete files since I keep printing test parts and I don't need them anymore on the SD card.

[image]

bambu123, 2024-02-06:
> Same problem here... The latest P1P/P1S update stated the following: "Support SD card remote browsing for time-lapses and models"
> 
> Does this only apply when using the cloud? If so, this should be made clear in the release notes and please be released for the lan-mode as well. Thanks in advance!

Yes, LAN-Only mode file protocol is not support for all printer models until now.

ian-morgan99, 2025-02-01:
Same problem on a brand new A1 currently. I would really like to be able to access the SD card in lan only mode.
This could be done via FTPS over port 990, as I have confirmed manually today.