bambulab/BambuStudio #3211: Intermittend connection problems with Bambu Studio 1.8.2

Hen3V, 2023-12-23:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home

### Printer

X1C

### How to reproduce

Goto 'Device' page.

### Actual results

The connection with the printer is intermittent lost. I am able to send files to the printer. Also, Bamby Handy has no connection issues whatsoever, just Studio.

### Expected results

Expected are not connection issues.
And after rolling back to Studio version 1.7.7.89 the issue is gone, so it seems to be an issue in Studio.

Please note that my printer does not have a strong WLAN connection, which may be of influence. But again, this was never een issue before, and is not an issue with the older Bambu Studio or Bambu Handy.

### Project file & Debug log uploads

[Logs.zip](https://github.com/bambulab/BambuStudio/files/13759962/Logs.zip)
[project.3mf.zip](https://github.com/bambulab/BambuStudio/files/13759964/project.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

murdrrdbypirates, 2023-12-24:
I have been having the same issue since changing to Version 1.8.2, I also rolled back to v01.07.07.89 and the issue has disapeared.
I have a strong WIFI connection, my printer sits apprx. 1m from my router.

walterwongbbl, 2023-12-25:
Hi. @murdrrdbypirates 
Please take a look at this
https://github.com/bambulab/BambuStudio/issues/3122

murdrrdbypirates, 2023-12-25:
> Hi. @murdrrdbypirates Please take a look at this #3122

Thanks very much for the info. I've synced up/resynced my Windows time again and installed the new version. All is working as expected.

walterwongbbl, 2023-12-26:
well done

closed 2023-12-26 by walterwongbbl