bambulab/BambuStudio #3213: Coloring not working on components

Tomzrk, 2023-12-24:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

window 11

### Additional system information

Intel(R) Core(TM) i7-1065G7 CPU @ 1.30GHz   1.50 GHz
12.0 GB  RAM

### Printer

bambu lab p1s

### How to reproduce

I designed the part on fusion 360 and exported the file stp , opened it on bambu studio and tried coloring some components but after slicing the part only the body show up without the colored components
[image]

### Actual results

[image]
only body show up without components

### Expected results

i done the same thing on different part and it worked fine 

### Project file & Debug log uploads

[xbox cont bug.zip](https://github.com/bambulab/BambuStudio/files/13760591/xbox.cont.bug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-12-28:
[image]
Because the Object of this model contains multiple components, and there is overlap between the components. In the overlapping part, one color is 1 and the other is 4. When two colors overlap, we give priority to displaying the color with the smaller number. So if you want to show the desired effect, you can add a new material 5 and use material 5 to color the bottom surface, as shown in the picture.

Tomzrk, 2024-01-03:
How can I print it that way ?

zhimin-zeng-bambulab, 2024-01-12:
You can create a new filament 5, and then repaint all the colors of the original filament 1 with filament 5. Then when sending for printing, just map filament 5 to the corresponding ams slot.

zhimin-zeng-bambulab, 2024-02-04:
@Tomzrk  Did the above method solve your problem?

Tomzrk, 2024-02-12:
No it didn't

zhimin-zeng-bambulab, 2024-02-18:
@Tomzrk  After you add a new filament 5, the bottom surface still only has one color after slicing?
[image]

zhimin-zeng-bambulab, 2024-03-08:
@Tomzrk  Could you please tell us what problems currently exist with the above method?

closed 2024-06-26 by zhimin-zeng-bambulab