bambulab/BambuStudio #3215: Bambu Studio does not open

SteveJohnny99, 2023-12-24:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11 Pro 64bit, 23H2

### Additional system information

AMD Ryzen 7 5800G
32GB DDR4 3600MHz RAM
RX580 GPU
1TB M.2 SSD

### Printer

Bambu Lab P1S

### How to reproduce

I have been working with Bambu Studio until 1.7.7.89 without problems, then I tried updating to 1.8.0 and the program never opened again.
When I click on the icon the white square appears and tells "Bambu studio - loading configuration". The program remeins stuck there and uses 0% CPU, so after some hours I switched it off.
I tried many versions from 1.7 to 1.8.2.57 (found on github to solve another error), tried to delete every folder possible after uninstalling the old and before installing a new version, tried various things I found on forums (eg. reinstalling Windows Media Player), Windows compatibility issues detection, nothing worked. Every version stucks at "loading configuration" before even displaying the fullscreen.

### Actual results

On my laptop (an old Lenovo with Windows 11 Pro 21H2) I still have the 1.7.7.89 version running so I tried waiting some weeks and downloading the updated versions that came out without success.
I am currently using Solidworks on my PC and Bambu Studio on my laptop, but since SW doesn't run on the laptop I need Bambu Studio on my PC.
I really have no clue about what's blocking me on configuration loading but since it's been a while I'm asking for help.
Note that I have no debug/log/anything because the program doesn't even start.

### Expected results

.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-12-25:
The files in the user configuration directory may be damaged. Try renaming the folder %APPDATA%\BambuStudio and reinitializing.

If it still doesn't work, please refer to the wiki below, collect logs and send them to us.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
> Windows Platform > Freeze Problem

SteveJohnny99, 2023-12-25:
Thank you for the reply. I tried renaming the folder but if I set a name different from " BambuStudio" a new folder is automatically recreated, and if I rename it as it is, nothing changes.
I read the page Windows platform/freeze problem in the link you sent me and created the .DMP file.
I uploaded it on my google drive in this folder: https://drive.google.com/drive/folders/1__KaNT7WNpEd_r6po94_nUFetbdQ_WPp?usp=drive_link

bambu123, 2023-12-28:
> Thank you for the reply. I tried renaming the folder but if I set a name different from " BambuStudio" a new folder is automatically recreated, and if I rename it as it is, nothing changes. I read [...]

Are you using a HDMI display?

SteveJohnny99, 2024-01-02:
Hi, I read about display problems with the program so I tried DP, HDMI and DVI display on both integrated and external graphics. I am currently using a DVI vertical monitor and an HDMI display with Display Port to HDMI connector together as it didn't seem to change anything.

ShadowRiver715, 2024-01-02:
@SteveJohnny99 did you fix your slicer

ShadowRiver715, 2024-01-02:
> The files in the user configuration directory may be damaged. Try renaming the folder %APPDATA%\BambuStudio and reinitializing.
>
> If it still doesn't work, please refer to the wiki below, collect [...]

I can not find the folder for %APPDATA%\BambuStudio

Haidiye01, 2024-01-03:
[image]
 like this

SteveJohnny99, 2024-01-13:
[debug_Sat_Jan_13_10_50_55_8568.log.zip](https://github.com/bambulab/BambuStudio/files/13927586/debug_Sat_Jan_13_10_50_55_8568.log.zip)

Don't know if this can help, here is the .log.0 file I found in the %APPDATA%\BambuStudio folder today after a new reinstalling.

ShadowRiver715, 2024-01-22:
It was a driver issue for me idk if it is for yall i just reinstalled my gpu drivers

MarkJeronimus, 2025-01-02:
In my case it was because some 'last*'and 'history' paths pointed to a network disk, but Windows' network SMB service had crashed and every network access hung forever interruptibly.

Haidiye01, 2025-01-02:
ok,i have received your letter ,thank you!

Haidiye00, 2025-11-19:
https://github.com/bambulab/BambuStudio/releases/tag/v02.04.00.70
Hello, the new version has been released, please update.
If there are any further questions, let's continue the discussion. Thank you