bambulab/BambuStudio #3216: Cannot open STEP file, never stop opening.

Supraconductor, 2023-12-24:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1

### How to reproduce

Open the STEP file - will never stop trying to open. 

### Actual results

Opening process never terminate - cannot open the STEP file. 

### Expected results

Just open it.

### Project file & Debug log uploads

[Head.zip](https://github.com/bambulab/BambuStudio/files/13762587/Head.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MklBambu, 2023-12-25:
[image]
This file can be opened, but it will take some time.

SaltWei, 2024-01-22:
[image]
Very slow to load this step file

closed 2025-04-29 by MackBambu