bambulab/BambuStudio #3218: INCORRECT PASSWORD WHEN PASSWORD HASN'T CHANGED

TennisRunner, 2023-12-25:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

p1p

### How to reproduce

1. Start BS.
2. Slice something
3. Print plate

### Actual results

[image]

Says that its syncing 10-30 seconds and then does a popup that says "Incorrect Password".
Check the printer and the password hasn't changed.
Manually write down the password for the millionth time and type it in manually to the popup box that doesn't save the last password.

### Expected results

It shouldn't fail and say the password is incorrect.
It shouldn't erase the last password and require you to manually walk over to the printer to copy it again.
How convenient that for months the lan only mode that doesn't leak your models to BS's database has a randomly occurring nag screen that hasn't ever been fixed.

### Project file & Debug log uploads

[Untitled.zip](https://github.com/bambulab/BambuStudio/files/13763855/Untitled.zip)
[debug_Sun_Dec_24_22_15_07_28488.log.zip](https://github.com/bambulab/BambuStudio/files/13763864/debug_Sun_Dec_24_22_15_07_28488.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-12-25:
Hi   @TennisRunner 
Have you modified the access code on the printer side? As far as I know, the printer will automatically change the access code in several scenarios

TennisRunner, 2023-12-25:
> Hi @TennisRunner Have you modified the access code on the printer side? As far as I know, the printer will automatically change the access code in several scenarios

Haven't changed it in over a week.
I have most of it memorized by now with the amount of times I've had to do this.

walterwongbbl, 2023-12-25:
@TennisRunner 
This should be a bug, we will fix it

walterwongbbl, 2024-01-18:
Hi @TennisRunner 
How long does it take for the access code to prompt as incorrect? We tried to reproduce this issue but were unsuccessful. Is there any additional action required? Could you please provide the steps to reproduce the problem?

TennisRunner, 2024-01-18:
I updated to the latest firmware on the printer and the issue doesn't seem to be coming back. 
I'll re-open this if ever does. 
Thanks.

closed 2024-01-18 by TennisRunner

builder555, 2024-12-17:
this is happening again on the latest firmware