bambulab/BambuStudio #3219: Custom settings passed to the CLI directly throw the help menu with no signs of error message

THE-SIMPLE-MARK, 2023-12-25:
### Bambu Studio Version

BambuStudio-01.08.01.57

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 22.04.3 LTS

### Printer

Bambu Lab P1P

### How to reproduce

1. Get the linux version of BambuStudio
2. Run the following command:
```shell
"./utils/bambuStudio/bin/bambu-studio" --debug 5 --outputdir './tmp/sliceOutputs/xp83kg7ni8133g1jr7ohbaky/' --ensure-on-bed --load-settings 'utils/slicerConfigs/machine/Bambu Lab P1P 0.4 nozzle.json;utils/slicerConfigs/process/0.20mm Standard @BBL P1P.json' --load-filaments 'utils/slicerConfigs/filament/Bambu PLA Basic @BBL P1P.json' --slice 0 --info --enable_support="true" "./tmp/fileUploads/xp83kg7ni8133g1jr7ohbaky.stl"
``` 
The STL can be anything, as well as the output directory. Filament and machine/process settings are irrelevant as well.

### Actual results

I get thrown the help menu with no actual info about what has gone wrong.
```shell
[2023-12-23 13:36:50.662112] [0x00007fb73d9d6c80] [trace]   Initializing StaticPrintConfigs
BambuStudio-01.08.01.57:
Usage: bambu-studio [ OPTIONS ] [ file.3mf/file.stl ... ]

OPTIONS:
 --allow-multicolor-oneplate
...
``` 

### Expected results

The command runs as expected without any issues.

### Project file & Debug log uploads

The project file is irrelevant in this case, however these are the logs I get thrown when running the command mentioned above:
```shell
[2023-12-23 13:36:50.662112] [0x00007fb73d9d6c80] [trace]   Initializing StaticPrintConfigs
BambuStudio-01.08.01.57:
Usage: bambu-studio [ OPTIONS ] [ file.3mf/file.stl ... ]

OPTIONS:
 --allow-multicolor-oneplate
...
``` 
I truncated the help menu purpusefully because I think everyone gets the point.

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-12-26:
there is an option "enable_support" in the command line
please use '-' instead of '_'

--enable-support="true"

THE-SIMPLE-MARK, 2023-12-26:
> there is an option "enable_support" in the command line please use '-' instead of '_'
> 
> --enable-support="true"

Ah that's perfect. Can all editor option IDs be used as CLI args or only a select few? If so, can you tell me where these args are defined so I can account for them?

THE-SIMPLE-MARK, 2023-12-27:
@lanewei120 An answer to my question above would be greatly appreciated!

lanewei120, 2023-12-28:
> @lanewei120 An answer to my question above would be greatly appreciated!

yes, most of the option can be set under the CLI

you can find the system profiles in C:\Program Files\Bambu Studio\resources\profiles\BBL
and all the options can be found there

closed 2023-12-30 by THE-SIMPLE-MARK