bambulab/BambuStudio #3221: Application does not work

cjbeckjr, 2023-12-25:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

AMD Ryzen 7 5800U with Radeon Graphics            1.90 GHz
16.0 GB (15.4 GB usable)
 NO MICROSOFT BROWSERS Installed so if linked to Microsoft of any sort this makes your application useless

### Printer

ANY

### How to reproduce

Open application and nothing is displayed under Home or project

### Actual results

[image]

### Expected results

a populated screen with a supported browser not junk from crap company like Microsoft

### Project file & Debug log uploads

NA
[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-12-27:
Please refer: 
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/install-issue
Blank Home page and blank Wizard Dialog (Windows Platform)

SaltWei, 2024-01-18:
Hello@cjbeckjr Is this wiki useful for solving the issue?