bambulab/BambuStudio #3223: Camera view not working [Linux]

ashermyers, 2023-12-25:
### Bambu Studio Version

Latest Stable

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 20.04 LTS 64 bit

### Printer

P1S

### How to reproduce

When printing, attempt to view the internal printer camera.

### Actual results

App Crashes.

### Expected results

Shows live feed from camera.

### Project file & Debug log uploads

[logs.zip](https://github.com/bambulab/BambuStudio/files/13767405/logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-12-26:
Do you have gstreamer installed?

ashermyers, 2023-12-28:
Yes

bambu123, 2023-12-28:
> Yes

Can you help me call practice run with gdb in the console to get more error messages?

ashermyers, 2023-12-28:
Yes, can you provide instructions?

bambu123, 2024-01-24:
> Yes, can you provide instructions?

Sorry for the late reply. Has your problem been resolved?

To do:

$ gdb Bambu_Studio_linux_ubuntu_vXXX.AppImage
run
bt

please input the 'run' command in gdb. then when it stops on a crash, input the 'bt' command.
then send the console output back here.

AbeNaws, 2024-06-13:
@bambu123 I ran the first command, and got this message: 

$ gdb Bambu_Studio_linux_ubuntu-v01.09.02.57.AppImage
GNU gdb (Debian 13.2-1+b1) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from Bambu_Studio_linux_ubuntu-v01.09.02.57.AppImage...
(No debugging symbols found in Bambu_Studio_linux_ubuntu-v01.09.02.57.AppImage)

AbeNaws, 2024-06-13:
here is the output from bt: 

(gdb) bt
#0  ____longjmp_chk ()
    at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:168
#1  0xee8777a6ad7683af in  ()
(gdb)

bambu123, 2024-06-14:
It seems to be an inconsistency issue with the JPEG library.
the same as https://github.com/bambulab/BambuStudio/issues/1814
and https://github.com/bambulab/BambuStudio/issues/1665#issuecomment-1550380137

could you have a check in the above issues?