bambulab/BambuStudio #3224: Translation Of BambuStudio_es.po to Spanish

Keyn11, 2023-12-25:
Hello, I was translating with poedit and I don't know how to upload it to the main folder, I am asking for help.
I don't know anything abot githu, I'm a newbie.
It won't let me upload the files in any way, neither zip nor .po, my email is transwq@gmail.com

Keyn11, 2023-12-25:
Please rename to bambustudio.mo and bambustudio.po.
[image]
[image]

DanBao-Bambu, 2023-12-26:
Thank you very much for everything you have done to Bambu Studio, but Spanish translation is currently maintained by Bambu official. Please refer to this wiki, https://wiki.bambulab.com/en/bambu-studio/contribute-languages . We will regularly update the translated content. If there are any translations that you feel are incorrect or could be better, you can raise them here and we will provide feedback on your suggestions. Thank you again for your efforts.