bambulab/BambuStudio #3225: Setting to keep layer time constant across objects when printing with By Layer print sequence

bjfishman, 2023-12-26:
**Is your feature request related to a problem? Please describe.**
When printing multiple objects with different heights on a single plate in By Layer sequence (see screenshot 1), the layer time of tall objects can become smaller after all layers of the shorter objects are completed (see screenshot 2), because the total layer time for the tall parts is only the individual layer time of the tall object while the total layer time for all preceding layers is the individual layer times of all objects combined. This can result in an undesired change of surface finish for all layers of the tall object that are taller than the other objects (see screenshot 3).

**Describe the solution you'd like**
Implement a setting that allows me to force a constant layer time across all objects. When enabled, the total layer time of all layers above the red line in screenshot 2 should be identical to the total layer time of the preceding layers. This would result in an uniform surface finish for all parts, at the cost of printing time.

Ideally, implement a subsetting that allows me to control whether layers should be slowed down to match the longest total layer time, or sped up to match the shortest total layer time. In my case, this would allow me to choose between all objects having a matte finish (layers above red line slowed down) and all objects having a shiny finish (layers below the red line sped up).

**Describe alternatives you've considered**
At the moment, my options are:
- print the objects individually on separate plates, measure how long a layer of the short objects takes, use the measured time as min layer time for the tall objects (inconvenient, requires me to mess around with my filament profiles)
- manually add speed change G-code after the layer where the short objects finish printing (inconvenient, needs trial & error to figure out the correct speed to achieve desired surface finish)
- slow down the entire print to achieve an uniform matte finish (costs a lot of time)

**Additional context**
[image]
[image]
[image]

DuckTown407, 2024-01-01:
You could do that yourself in 2 different ways. In the filament settings use the minimum layer time option and find the layer time *before* the smaller sections are finished. Set the minimum layer time to that and then the layers when the shorter sections are done will be as long as before. OR use a height range modifier and slow the speeds down after the shorter sections are finished.

kvnper, 2024-02-29:
I proposed a solution here:

https://github.com/bambulab/BambuStudio/issues/3657