bambulab/BambuStudio #3229: Unable to change filament at layer height when first layer has text with a filament color change.

mturley777, 2023-12-27:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1.  On any print ready to slice, add text to the bottom of the print (embedded in the first layer).
2.  Change the filament color on the text object created above to another color.
3.  Now slice the model.  Then select a layer in the sliced model using the layer slider.
4.  Now, right click on the selected layer and attempt to change filament at that layer.

### Actual results

When you try to change filament at layer, it is greyed out if you have the text along with a filament change on the first layer.  If you have no text on the first layer, or the text on the first layer does not have a filament change, then the change filament at layer is not greyed out.

Example - Text on first layer with filament change:

[image]

Then result when trying to add a filament change at layer:

[image]

### Expected results

I expect that I can change filament at any given layer while also having text with a filament color change on the first layer.

### Project file & Debug log uploads

[Filament change at layer broken.zip](https://github.com/bambulab/BambuStudio/files/13775571/Filament.change.at.layer.broken.zip)

[Filament change at layer works fine.zip](https://github.com/bambulab/BambuStudio/files/13775573/Filament.change.at.layer.works.fine.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

LiZ-Li-BBL, 2024-01-16:
If you want to change filament at any given layer while also having text with a filament color change on the first layer, you should use Height Range Modifier or Color Painting Tool.

LiZ-Li-BBL, 2024-01-16:
The "Change Filament" function on the layer slider actually only works if the entire plate has only one color.

mturley777, 2024-01-16:
Sounds like a great opportunity to enhance the app to be more flexible then!

closed 2024-01-16 by mturley777

JansenSmith, 2024-11-13:
Fixing this issue by having the Change Filament be greyed out if there are multiple colors on nearby layers would be a great improvement to my workflow.