bambulab/BambuStudio #3232: BambuStudio Slicer issue

Sembach-Mabuhay, 2023-12-27:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS, Ventura 13.6.3

### Additional system information

CPU: 2.9 GHz Quad-Core Intel Core i7
Graphics: Radeon Pro 560 4 GB Intel HD Graphics 630 1536 MB
Memory: 16 GB 2133 MHz LPDDR3

### Printer

Bambu Labs X1 Carbon with AMS

### How to reproduce

1. Click slice on any project
2. click preview
3. 

### Actual results

I don't get an image of what's being sliced.

[image]
[image]
[image]

### Expected results

I should be able to see the slicer and use it.  yet the slider or picture of the file never shows up.

### Project file & Debug log uploads

[CrystalDragon_NewLegs_Curled.3mf.zip](https://github.com/bambulab/BambuStudio/files/13779748/CrystalDragon_NewLegs_Curled.3mf.zip)
[Archive.zip](https://github.com/bambulab/BambuStudio/files/13779831/Archive.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sanctas, 2023-12-27:
Looks like you have multiple plates in this file, you might be viewing an empty plate. Select the plate with the dragon on it and then select "slice plate" from the drop down in the top right (currently it says "Slice All"). I think that will generate the preview for you.

Sembach-Mabuhay, 2023-12-27:
Ok, i just did that.  And i got the exact same result.  I can print the project, but the preview window is still blank.

Sembach-Mabuhay, 2023-12-27:
Thanks for the fast reply.

I actually uninstalled the Bambu Studio, and re-installed it.  And now everything is working as expected.

Thanks,
J

closed 2024-01-17 by SaltWei