bambulab/BambuStudio #3233: eSun PLA+ System Preset bad volumetric speed for 0.2 nozzle

alienm16, 2023-12-27:
### Bambu Studio Version

1.8.0.62

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 23h2 build 22631.2715

### Printer

Bambu Lab X1CC with firmware 1.07.07.01

### How to reproduce

Select printer Bambu Lab X1C with 0.2 nozzle
Apply the eSUN PLA+ system preset to a roll
Edit the filament and you see that volumetric speed is set to 1

For the 0.4 nozzle, the volumetric speed is 16

### Actual results

Print with eSUN PLA+ default system preset take longer to print

### Expected results

Print faster with the good max volumetric speed

### Project file & Debug log uploads

[eSUN PLA+ @BBL X1C 0.2 nozzle.json](https://github.com/bambulab/BambuStudio/files/13780369/eSUN.PLA%2B.%40BBL.X1C.0.2.nozzle.json)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Xiuqi-Huang-CN, 2023-12-28:
@alienm16 
Hello. We have retested the max volumetric speeds of several kinds of PLA, including Bambu PLA Basic, PLA Matte and eSun PLA+ and etc., and the most of them can reach about 3 mm³/s. And to mke sure that the qualities of most prints will  be high enough, we will adjust them to be 2 mm³/s by default in the next version. In your printing cases, if higher printing speeds and shorter printing times are more important, you can try 2.2, 2.5 and maybe 3.0 mm³/s respectively and check which works out as your expectation.
Hope that this can help you.

alienm16, 2024-01-08:
> @alienm16 Hello. We have retested the max volumetric speeds of several kinds of PLA, including Bambu PLA Basic, PLA Matte and eSun PLA+ and etc., and the most of them can reach about 3 mm³/s. And to [...]

I have made a test this weekend with Orca and max flow test and the bambu pla matte dark red and I have reach 13-14 mm3/s and I have set 12 in the filament profile and my print was good and rather than 1d and 17 hours, the print was only 24 hours, a very big difference.