bambulab/BambuStudio #3234: Bambu Studio Crashes after latest network driver update.

justpie, 2023-12-27:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu  22.04

### Printer

X1 Carbon

### How to reproduce

Updating to latest network driver causes my BambuStudio to segfault when launching.

### Actual results

When I launch BambuStudio when not logged in works fine, I just can't use remote printer settings. However, when i login to my account bambu studio crashes immediately after launching. Terminal Error reports Seg Fault. 

### Expected results

It crashes../

### Project file & Debug log uploads

2023-12-27 15:08:01.399325[Thread 0x00007fe8c1313c80]:gui mode, Current BambuStudio Version 01.08.02.56
2023-12-27 15:08:01.403940[Thread 0x00007fe8c1313c80]:load_language: language , initial: 1
2023-12-27 15:08:01.403960[Thread 0x00007fe8c1313c80]:language provided by BambuStudio.conf: en_US
2023-12-27 15:08:01.404067[Thread 0x00007fe8c1313c80]:Switching wxLocales to en_US
2023-12-27 15:08:01.413198[Thread 0x00007fe8c1313c80]:Failed to load to hints.cereal. File does not exists. /home/davey/.config/BambuStudio/user/hints.cereal
2023-12-27 15:08:01.415404[Thread 0x00007fe8c1313c80]:Update_dark_mode_flag: switch the current dark mode status to 1 
2023-12-27 15:08:01.417826[Thread 0x00007fe8c1313c80]:begin to show the splash screen...
2023-12-27 15:08:01.524529[Thread 0x00007fe8c1313c80]:loading systen presets...
2023-12-27 15:08:01.530389[Thread 0x00007fe8c1313c80]:[BBL Updater]:Checking whether the profile from resource is newer
2023-12-27 15:08:01.540776[Thread 0x00007fe8c1313c80]:initialize_network_module, successfully loaded library /home/davey/.config/BambuStudio/plugins/libbambu_networking.so, module 0x5623b6fccce0
2023-12-27 15:08:01.540827[Thread 0x00007fe8c1313c80]:get_network_function, can not find function bambu_network_get_profile_3mf
2023-12-27 15:08:01.540837[Thread 0x00007fe8c1313c80]:on_init_network: on_init_network, load dll ok
2023-12-27 15:08:01.541797[Thread 0x00007fe8c1313c80]:get_network_agent_version=01.08.02.04
2023-12-27 15:08:01.541813[Thread 0x00007fe8c1313c80]:on_init_network: on_init_network, compatibility version
2023-12-27 15:08:01.552541[Thread 0x00007fe8c1313c80]:on_init_network, create network agent...
2023-12-27 15:08:01.553207[Thread 0x00007fe8c1313c80]:NetworkAgent, this 0x5623b6fb7a90, network_agent=0x5623b6fb7a70, create_agent_ptr=1, log_dir=/home/davey/.config/BambuStudio
2023-12-27 15:08:01.558349[Thread 0x00007fe8c1313c80]:init_networking_callbacks: enter, m_agent=0x5623b6fb7a90
2023-12-27 15:08:01.558422[Thread 0x00007fe8c1313c80]:init_networking_callbacks: exit, m_agent=0x5623b6fb7a90
2023-12-27 15:08:01.559937[Thread 0x00007fe8c1313c80]:load_presets enter, substitution_rule 3, preferred printer_model_id 
2023-12-27 15:08:01.559959[Thread 0x00007fe8c1313c80]:load_vendor_configs_from_json enter, path /home/davey/.config/BambuStudio/system, compatibility_rule 2
2023-12-27 15:08:01.560557[Thread 0x00007fe8c1313c80]:load_vendor_configs_from_json: parse /home/davey/.config/BambuStudio/system/BBL.json, got description:  "the initial version of BBL configurations"
2023-12-27 15:08:01.618510[Thread 0x00007fe8c1313c80]:load_user_presets521 entry and user is: 4003165649
2023-12-27 15:08:01.618542[Thread 0x00007fe8c1313c80]:load_presets enter, load presets from "/home/davey/.config/BambuStudio/user/4003165649/process", current type process
2023-12-27 15:08:01.619736[Thread 0x00007fe8c1313c80]:load_presets enter, load presets from "/home/davey/.config/BambuStudio/user/4003165649/filament", current type filament
2023-12-27 15:08:01.619861[Thread 0x00007fe8c1313c80]:load_presets enter, load presets from "/home/davey/.config/BambuStudio/user/4003165649/machine", current type machine
2023-12-27 15:08:01.619884[Thread 0x00007fe8c1313c80]:update_compatibility for all presets enter
2023-12-27 15:08:01.619904[Thread 0x00007fe8c1313c80]:update_compatibility for all presets exit
2023-12-27 15:08:01.619907[Thread 0x00007fe8c1313c80]:update_compatibility for all presets enter
2023-12-27 15:08:01.619914[Thread 0x00007fe8c1313c80]:update_compatibility for all presets exit
2023-12-27 15:08:01.620349[Thread 0x00007fe8c1313c80]:update_compatibility for all presets enter
2023-12-27 15:08:01.620527[Thread 0x00007fe8c1313c80]:update_compatibility for all presets exit
2023-12-27 15:08:01.620531[Thread 0x00007fe8c1313c80]:load_presets finished, returned substitutions 0
2023-12-27 15:08:01.620534[Thread 0x00007fe8c1313c80]:create the main window
2023-12-27 15:08:01.730458[Thread 0x00007fe8c1313c80]:CreateWebView: file:/tmp/.mount_BambuSTJAdKB/resources/web/homepage/index.html?lang=en_US
2023-12-27 15:08:01.889304[Thread 0x00007fe8c1313c80]:select_technology: change the printer technology from  to 
2023-12-27 15:08:01.891840[Thread 0x00007fe8c1313c80]:get_current_slice_result:m_current_plate 0, current_plate 0x5623b90774e0
2023-12-27 15:08:01.899861[Thread 0x00007fe8c1313c80]:set_current_panel: current_panel 0, new_panel 0x5623b8a36130
2023-12-27 15:08:02.128970[Thread 0x00007fe8c1313c80]:ImageGrid::SetStatus: No printers.
2023-12-27 15:08:02.130766[Thread 0x00007fe8c1313c80]:monitor: show_status = 4
2023-12-27 15:08:02.133868[Thread 0x00007fe8c1313c80]:market_scoring_hide hide market scoring page
[... 119 lines cut ...]
2023-12-27 15:08:11.861935[Thread 0x00007fe8c1313c80]:stop, enter
2023-12-27 15:08:11.863557[Thread 0x00007fe8c1313c80]:object_list_changed: can_slice 0, model_fits= 1, export_in_progress 0, has_printable_instances 0 
2023-12-27 15:08:11.863566[Thread 0x00007fe8c1313c80]:get_enable_print_status: m_print_select 1, enable= 0 
2023-12-27 15:08:11.863569[Thread 0x00007fe8c1313c80]:update_slice_print_status m_slice_select 1: can_slice= 0, can_print 1, enable_slice 0, enable_print 0 
2023-12-27 15:08:11.863597[Thread 0x00007fe8c1313c80]:load_current_preset: enter
2023-12-27 15:08:11.864455[Thread 0x00007fe8c1313c80]:put_other_changes
2023-12-27 15:08:11.864460[Thread 0x00007fe8c1313c80]:update_from_presetsproject file name is empty
2023-12-27 15:08:11.881680[Thread 0x00007fe8c1313c80]:load_current_preset: exit
2023-12-27 15:08:11.881694[Thread 0x00007fe8c1313c80]:load_current_preset: enter
2023-12-27 15:08:11.882427[Thread 0x00007fe8c1313c80]:put_other_changes
2023-12-27 15:08:11.882433[Thread 0x00007fe8c1313c80]:update_from_presetsproject file name is empty
2023-12-27 15:08:11.883331[Thread 0x00007fe8c1313c80]:load_current_preset: exit
2023-12-27 15:08:11.883338[Thread 0x00007fe8c1313c80]:load_current_preset: enter
2023-12-27 15:08:11.883438[Thread 0x00007fe8c1313c80]:reset: current result id 4294967295 
2023-12-27 15:08:11.885175[Thread 0x00007fe8c1313c80]:put_other_changes
2023-12-27 15:08:11.885183[Thread 0x00007fe8c1313c80]:update_from_presetsproject file name is empty
2023-12-27 15:08:11.886256[Thread 0x00007fe8c1313c80]:load_current_preset: exit
2023-12-27 15:08:11.888911[Thread 0x00007fe8c1313c80]:put_other_changes
2023-12-27 15:08:11.888920[Thread 0x00007fe8c1313c80]:update_from_presetsproject file name is empty
2023-12-27 15:08:11.888947[Thread 0x00007fe8c1313c80]:set_project_name7164 project is:Untitled

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-12-28:
@justpie 

how about using this image https://github.com/bambulab/BambuStudio/actions/runs/7334871417/artifacts/1135915919?

justpie, 2023-12-28:
This appears to be working. Thank you!

closed 2023-12-28 by justpie