bambulab/BambuStudio #3235: crash on start when logged in

tmdag, 2023-12-27:
### Bambu Studio Version

fedora_v01.08.02.56-20231213

### Where is the application from?

Bambu Lab github releases

### OS version

6.6.6-100.fc38.x86_64

### Additional system information

Host: 21HF001XUS ThinkPad P14s Gen 4
DE: GNOME 44.6 
CPU: 13th Gen Intel i7-1370P (20) @ 5.000GHz 
GPU: Intel Raptor Lake-P [Iris Xe Graphics] 
GPU: NVIDIA RTX A500 Laptop GPU 

### Printer

BambuLab X1 

### How to reproduce

downloaded official .Appimage for fedora
run ./BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage

### Actual results

first run works, asks for location, login
After login is provided, then application segfaults
Any new attempt of running the application again, segfaults.

### Expected results

not to segv

### Project file & Debug log uploads

```
(bambu-studio:1330025): Gtk-CRITICAL **: 12:19:36.143: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
[New Thread 0x7f2dc5fff6c0 (LWP 1330784)]
[New Thread 0x7f2dc57fe6c0 (LWP 1330785)]
[New Thread 0x7f2dc4ffd6c0 (LWP 1330787)]
[New Thread 0x7f2dbbfff6c0 (LWP 1330788)]
[Detaching after vfork from child process 1330786]
[Thread 0x7f2dbbfff6c0 (LWP 1330788) exited]
[Thread 0x7f2dc4ffd6c0 (LWP 1330787) exited]
[Detaching after vfork from child process 1330790]
[New Thread 0x7f2dc4ffd6c0 (LWP 1330792)]
[New Thread 0x7f2dbbfff6c0 (LWP 1330794)]
[Detaching after vfork from child process 1330793]
[Thread 0x7f2dbbfff6c0 (LWP 1330794) exited]
[New Thread 0x7f2dbbfff6c0 (LWP 1330796)]
[New Thread 0x7f2dbb7fe6c0 (LWP 1330797)]
[New Thread 0x7f2dbaffd6c0 (LWP 1330798)]
[Thread 0x7f2dbaffd6c0 (LWP 1330798) exited]
[New Thread 0x7f2dbaffd6c0 (LWP 1330800)]
[Thread 0x7f2dbaffd6c0 (LWP 1330800) exited]
[Thread 0x7f2dc57fe6c0 (LWP 1330785) exited]
[New Thread 0x7f2dc57fe6c0 (LWP 1330801)]
[Thread 0x7f2dc57fe6c0 (LWP 1330801) exited]
[Thread 0x7f2dc4ffd6c0 (LWP 1330792) exited]
[New Thread 0x7f2dc4ffd6c0 (LWP 1330802)]
[New Thread 0x7f2dc57fe6c0 (LWP 1330803)]
[New Thread 0x7f2dbaffd6c0 (LWP 1330804)]
[New Thread 0x7f2dba7fc6c0 (LWP 1330805)]
[Thread 0x7f2dba7fc6c0 (LWP 1330805) exited]

Thread 553 "bambustu_main" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f2dbaffd6c0 (LWP 1330804)]
0x00007f309c6bf1b5 in __GI___libc_free (mem=0x2) at malloc.c:3344
3344      if (chunk_is_mmapped (p))                       /* release mmapped memory. */                                                                                                                                                                                                                                       
Missing separate debuginfos, use: dnf debuginfo-install xorg-x11-drv-nvidia-libs-545.29.06-1.fc38.x86_64
(gdb) where
#0  0x00007f309c6bf1b5 in __GI___libc_free (mem=0x2) at malloc.c:3344
#1  0x0000000003aa4f5e in ERR_STATE_free.part.0 ()
#2  0x0000000003acdfd8 in init_thread_stop.part ()
#3  0x0000000003ace013 in init_thread_destructor ()
#4  0x00007f309c6ab840 in __GI___nptl_deallocate_tsd () at nptl_deallocate_tsd.c:73
#5  __GI___nptl_deallocate_tsd () at nptl_deallocate_tsd.c:22
#6  0x00007f309c6ae7ba in start_thread (arg=<optimized out>) at pthread_create.c:455
#7  0x00007f309c734860 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) list
3339	
3340	  int err = errno;
3341	
3342	  p = mem2chunk (mem);
3343	
3344	  if (chunk_is_mmapped (p))                       /* release mmapped memory. */
3345	    {
3346	      /* See if the dynamic brk/mmap threshold needs adjusting.
3347		 Dumped fake mmapped chunks do not affect the threshold.  */
3348	      if (!mp_.no_dyn_threshold
```
[debug_Wed_Dec_27_12_50_21_1345146.log.txt](https://github.com/bambulab/BambuStudio/files/13781515/debug_Wed_Dec_27_12_50_21_1345146.log.txt)
[tutk_log_avapi_1345146.txt](https://github.com/bambulab/BambuStudio/files/13781516/tutk_log_avapi_1345146.txt)
[tutk_log_iotc_1345146.txt](https://github.com/bambulab/BambuStudio/files/13781517/tutk_log_iotc_1345146.txt)
[tutk_log_global_1345146.txt](https://github.com/bambulab/BambuStudio/files/13781518/tutk_log_global_1345146.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-12-28:
could you try this image https://github.com/bambulab/BambuStudio/actions/runs/7334871417/artifacts/1135915919 firstly?

we will upload a new fedora image soon

tmdag, 2023-12-28:
> could you try this image https://github.com/bambulab/BambuStudio/actions/runs/7334871417/artifacts/1135915919 firstly?
> 
> we will upload a new fedora image soon

Thanks! getting same result with this build unfortunately :(

[Thread 0x7ffd06ffd6c0 (LWP 11900) exited]

Thread 557 "bambustu_main" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffd19fff6c0 (LWP 11899)]
0x00007ffff04bf1b5 in __GI___libc_free (mem=0x2) at malloc.c:3344
3344      if (chunk_is_mmapped (p))                       /* release mmapped memory. */                                                                                                                                                                                                                                       
Missing separate debuginfos, use: dnf debuginfo-install xorg-x11-drv-nvidia-libs-545.29.06-1.fc38.x86_64
(gdb) trace
Tracepoint 1 at 0x7ffff04bf1b5: file malloc.c, line 3344.
(gdb)

lanewei120, 2023-12-28:
@tmdag 

hello, we just uploaded a new one here https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage

please use this one

thanks

tmdag, 2023-12-28:
thanks! testing!

Unfortunately still getting same error with this build. Is there any arg I can run to help debugging ?

`[Thread 0x7ffd157fa6c0 (LWP 66820) exited]
[New Thread 0x7ffcfffff6c0 (LWP 66822)]
[Thread 0x7ffcfffff6c0 (LWP 66822) exited]
[New Thread 0x7ffcfffff6c0 (LWP 66823)]

Thread 556 "bambustu_main" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffd14ff96c0 (LWP 66821)]
0x00007ffff04bf1b5 in __GI___libc_free (mem=0x2) at malloc.c:3344
3344      if (chunk_is_mmapped (p))                       /* release mmapped memory. */                                                                                                                                                                                                                                       
Missing separate debuginfos, use: dnf debuginfo-install xorg-x11-drv-nvidia-libs-545.29.06-1.fc38.x86_64
(gdb) p mem
$1 = (void *) 0x2
(gdb) list
3339	
3340	  int err = errno;
3341	
3342	  p = mem2chunk (mem);
3343	
3344	  if (chunk_is_mmapped (p))                       /* release mmapped memory. */
3345	    {
3346	      /* See if the dynamic brk/mmap threshold needs adjusting.
3347		 Dumped fake mmapped chunks do not affect the threshold.  */
3348	      if (!mp_.no_dyn_threshold
(gdb) x/i $pc
=> 0x7ffff04bf1b5 <__GI___libc_free+37>:	mov    -0x8(%rdi),%rax
(gdb) bt full
#0  0x00007ffff04bf1b5 in __GI___libc_free (mem=0x2) at malloc.c:3344
        ar_ptr = <optimized out>
        p = <optimized out>
        err = <optimized out>
#1  0x00007fffeab68022 in err_clear (deall=1, i=0, es=0x7ffcf800ec20) at crypto/err/err_local.h:86
        i = 0
        state = 0x7ffcf800ec20
#2  ERR_STATE_free (s=0x7ffcf800ec20) at crypto/err/err.c:209
        i = 0
        state = 0x7ffcf800ec20
#3  ERR_STATE_free (s=0x7ffcf800ec20) at crypto/err/err.c:202
        state = 0x7ffcf800ec20
#4  err_delete_thread_state (unused=<optimized out>) at crypto/err/err.c:651
        state = 0x7ffcf800ec20
#5  0x00007fffeabb77dc in init_thread_stop (arg=0x0, hands=0x7ffcf80025b0) at crypto/initthread.c:336
        curr = 0x7ffcf800efb0
        prev = 0x0
        tmp = <optimized out>
        gtr = 0x7ffd0c0059d0
#6  0x00007fffeabbc9df in init_thread_stop (hands=0x7ffcf80025b0, arg=0x0) at crypto/initthread.c:317
#7  init_thread_destructor (hands=0x7ffcf80025b0) at crypto/initthread.c:196
#8  0x00007ffff04ab840 in __GI___nptl_deallocate_tsd () at nptl_deallocate_tsd.c:73
        data = <optimized out>
        inner = <optimized out>
        level2 = 0x7ffcf800c370
        idx = <optimized out>
        cnt = 2
        self = <optimized out>
#9  __GI___nptl_deallocate_tsd () at nptl_deallocate_tsd.c:22
        self = <optimized out>
#10 0x00007ffff04ae7ba in start_thread (arg=<optimized out>) at pthread_create.c:455
        pd = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737224828496, 2191730678374425412, 140724955748032, -41560, 11, 140737488321536, 2191730678353453892, 2193075003190895428}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#11 0x00007ffff0534860 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
`

lanewei120, 2023-12-28:
my fault
using the wrong link, should be this one
https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_fedora_v01.08.02.56-20231228.AppImage

tmdag, 2023-12-28:
yay! works! didn't crash! thank you!

closed 2023-12-28 by tmdag