bambulab/BambuStudio #3236: "Creating a new project, unsaved changes popup" - changes for clarity.

rsilvers129, 2023-12-27:
I don't find this wording intuitive because it is calling your current modified settings "new settings" when they are existing/current settings while the user is taking an action to load a new project. It is calling the settings in the new project the "old" settings. 

I think it would be best to rethink this popup from scratch. What exactly it should say, I am not sure - but many people have this confusion as evidenced from this Facebook discussion:

https://www.facebook.com/groups/bambulabofficialusers/posts/1112404219703596

[image]

arekm, 2024-01-22:
@lanewei120 @sodachen22 @walterwongbbl could you look at this?

It's extremely confusing all the time even after a few months of Bambu Studio usage. And many other people have the same problem, see discussions here:

https://www.facebook.com/groups/bambulabofficialusers/posts/1112404219703596
or freshly
https://forum.bambulab.com/t/confusing-terminolgy-in-bambu-studio-transfer-vs-discard/51274
or older
https://www.reddit.com/r/BambuLab/comments/199px7d/bambu_studio_transfer_dialog/

There are few alternative suggestions in these discussions.

Rephrasing these shouldn't be a big job but would help a lot.

lanewei120, 2024-01-23:
thanks, we will improve this logic in later versions

DanBao-Bambu, 2024-03-08:
We attach great importance to your proposal and have carefully designed the description of the interface. This is a redesign of the interface, with the button text corresponding to the data description. Do you see if this interface is clear.
[image]

arekm, 2024-03-08:
It still confuses me. Like what's with the word "transfer"? It has the same meaning like "Use" in this context right? But different wording was chosen for "modified value" due to some reason and that's what confuses me.

So currently there is:

"You have previously modified your settings and are about to overwrite them with the new ones.
Do you want to keep your current modified settings, or use preset settings?"

I would change that sentence to something like:

"The settings you have modified in this session (Modified Value) are different from the stored configuration settings (Preset Value). Would you like to keep your current modifications?"

"Transfer Modified Value" -> "Use Modified Value". Looks simpler to me and doesn't cause confusion like what's special with "Transfer" used in this button in comparison to just "Use" used in the other button. Is "transfer" somehow different from "use", has special behavior? etc, etc

"Use Preset Value", that's ok
"Save Modified Value", ok

I would also change order of preset and modified columns, so column "Modified Value" would be on left (to match "Use Modified Value" button which would be on left) and column "Preset Value" would be on right (just like "Use Preset Value" is on right).

cmikel64, 2024-03-14:
How is this still not fixed? No one knows what to select on this screen.

SteveAtSmartersoft, 2024-03-17:
I hate this dialog, thanks for considering the fix!

If you don't mind a bit of input from someone 20+ years in IT;

- State the problem at the top - but don't ask a question.
- Then list the differences
- finally ask the question with straight choices

Just make sure the references are verbatim the same.   This what you have here is sadly one of the unforgivable mistakes in UI/UX.
[image]

cmikel64, 2024-03-17:
Lol. I could see this example being used in a UX/UI 101 textbook one day.

DanBao-Bambu, 2024-03-20:
The new interaction design has been completed, with the main changes described at the top of the dialog, button order, and description.
When loading a project, no modified value operation was used.
[image]
Keep the discard and use modified values when opening a new project.
[image]

cmikel64, 2024-03-20:
A swing and a miss.

SteveAtSmartersoft, 2024-03-20:
I reckon it's much better.

There are only two choices, and the wording is a lot clearer.   You keep it or you drop it.

I think the wording of the statement at the top could be polished a little, since it talks about "settings" plus "preset" - and what a "preset" is is undefined and probably irrelevant here.

I think dropping the "preset" entirely would be beneficial, since the screenshots are about 'settings' as well, change the headings to say "saved value" and "modified (unsaved) value" - something like this.

Then on load the statment:
"You've modified some settings (below).  Please review if you want to keep them or not."

And then "discard" on the left (both cases, gray) and "save" or "use" on the right (both cases, green)

arekm, 2024-08-06:
Improvements were already implemented, so this can be closed.

closed 2024-08-08 by lanewei120

nanodan, 2024-12-07:
This is still very confusing. I've saved a project, I go to close the window, and suddenly there's a popup that I've modified settings, do I want to save or discard? I just saved the project, so what is this asking me? Save what exactly, and save it where? Am I creating some new preset? Am I saving something to the project? Am I overwriting some default? This is not a clear design at all.

I click discard, and nothing really happens, okay, great, I guess? Out of curiosity I click save, ok now it wants me to save the process. But it was just asking me about settings, now it's talking about a process. Okay and it's asking me to save this process as a User Preset or Inside the Project, but I already saved the project, so the process doesn't stay with the project? It has to be saved separately? But can be global or local?

I'm probably going to use this as an example for our UX/UI team on how not to design an interface.