bambulab/BambuStudio #3237: Freeze on wizzard screen

tmdag, 2023-12-27:
### Bambu Studio Version

fedora_v01.08.02.56-20231213

### Where is the application from?

Bambu Lab github releases

### OS version

6.6.8-200.fc39.x86_64

### Additional system information

OS: Fedora Linux 39 (Workstation Edition) x86_64 
Host: Z9PE-D8 WS 1.0x 
Kernel: 6.6.8-200.fc39.x86_64 
CPU: Intel Xeon E5-2630 v2 (24) @ 3.100GHz 
GPU: NVIDIA GeForce GTX TITAN X 
NVIDIA-SMI 535.146.02   Driver Version: 535.146.02   CUDA Version: 12.2

### Printer

Bambu Lab X1

### How to reproduce

Downloaded BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage
./BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage

### Actual results

window opens and freezes with white content

### Expected results

not to freeze

### Project file & Debug log uploads

$HOME/.config/BambuStudio/log was empty.

** (bambu-studio:6025): CRITICAL **: 14:04:18.319: Cannot register URI scheme wxfs more than once

** (bambu-studio:6025): CRITICAL **: 14:04:18.319: Cannot register URI scheme memory more than once
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 1200x728: Invalid argument
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 1200x728: Invalid argument
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 1200x728: Invalid argument
Failed to create EGL images for DMABufs with file descriptors -1, -1 and -1
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 820x623: Invalid argument
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 820x623: Invalid argument
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 820x623: Invalid argument
Failed to create EGL images for DMABufs with file descriptors -1, -1 and -1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-12-28:
also please try this one https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage

tmdag, 2023-12-28:
Thank you @lanewei120 !

Tried this build on the machine mentioned in this ticket and it still freezes on the initial wizard screen

```
** (bambu-studio:23908): CRITICAL **: 23:44:02.407: Cannot register URI scheme wxfs more than once

** (bambu-studio:23908): CRITICAL **: 23:44:02.407: Cannot register URI scheme memory more than once
[Detaching after vfork from child process 24649]
[New Thread 0x7fff4dffb6c0 (LWP 24658)]
[New Thread 0x7fff4effd6c0 (LWP 24661)]
[New Thread 0x7fff327fc6c0 (LWP 24662)]
[Thread 0x7fff327fc6c0 (LWP 24662) exited]                                                                                                                                                                                                    
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 820x623: Invalid argument
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 820x623: Invalid argument
src/nv_gbm.c:99: GBM-DRV error (nv_gbm_bo_create): DRM_IOCTL_NVIDIA_GEM_ALLOC_NVKMS_MEMORY failed (ret=-1)

Failed to create GBM buffer of size 820x623: Invalid argument
Failed to create EGL images for DMABufs with file descriptors -1, -1 and -1
[New Thread 0x7fff327fc6c0 (LWP 24677)]
[Thread 0x7fff327fc6c0 (LWP 24677) exited]
[Thread 0x7fff4effd6c0 (LWP 24661) exited]
[Thread 0x7fff4ffff6c0 (LWP 24145) exited]
[Thread 0x7fffc59fd6c0 (LWP 24063) exited]
[New Thread 0x7fffc59fd6c0 (LWP 24737)]
[Thread 0x7fffc59fd6c0 (LWP 24737) exited]
[New Thread 0x7fffc59fd6c0 (LWP 24771)]
[Thread 0x7fffc59fd6c0 (LWP 24771) exited]
[New Thread 0x7fffc59fd6c0 (LWP 24831)]
^C
Thread 1 "bambustu_main" received signal SIGINT, Interrupt.
0x00007fffef325bcd in __GI___poll (fds=0xa53b250, nfds=8, timeout=1665) at ../sysdeps/unix/sysv/linux/poll.c:29
29        return SYSCALL_CANCEL (poll, fds, nfds, timeout); 
```

lanewei120, 2023-12-28:
@tmdag 
could you also upload the log files in ~/.config/BambuStudio/log to here

and could you also give a gif or some snapshots of your studio

bambu123, 2023-12-28:
@tmdag Please try the solution here https://github.com/bambulab/BambuStudio/issues/1828#issuecomment-1768736346

tmdag, 2023-12-28:
@bambu123 this solution worked!

`WEBKIT_DISABLE_COMPOSITING_MODE=1 ./BambuStudio_linux_fedora_v01.08.02.56-20231228.AppImage
`

bambu123, 2023-12-28:
Ok!

closed 2023-12-28 by bambu123