bambulab/BambuStudio #3238: Could NOT find GLEW (missing: GLEW_LIBRARIES) (found version "2.2.0")

tmdag, 2023-12-28:
### Bambu Studio Version

1.8.2

### Where is the application from?

Built from source by yourself or others

### OS version

6.6.8-100.fc38.x86_64

### Printer

Bambu Lab X1

### How to reproduce

$ sudo dnf instlall glew-devel.x86_64 glew-devel.x86_64

$ cmake ../ -DDESTDIR="/home/ats/Downloads/bambu/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1 -DJPEG_VERSION=6

$ cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="/home/ats/Downloads/bambu/BambuStudio_dep/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release

### Actual results

ALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 3
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: /home/ats/Downloads/bambu/BambuStudio_dep/usr/local (from cache or command line)
-- PROJECT_SOURCE_DIR: /home/ats/Downloads/bambu/BambuStudio
-- CMAKE_MODULE_PATH: /home/ats/Downloads/bambu/BambuStudio/cmake/modules/
-- LIBDIR: /home/ats/Downloads/bambu/BambuStudio/src
-- LIBDIR_BIN: /home/ats/Downloads/bambu/BambuStudio/build/src
-- Boost::boost exists
-- TBB found in /usr/lib64/cmake/TBB
OpenSSL include dir: /usr/include
OpenSSL libraries: /usr/lib64/libssl.so;/usr/lib64/libcrypto.so
-- Could NOT find EXPAT (missing: EXPAT_LIBRARY EXPAT_INCLUDE_DIR) 
-- FindGLEW: did not find GLEW CMake config file. Searching for libraries.
-- FindGLEW: GLEW_USE_STATIC_LIBS: ON.
-- FindGLEW: GLEW_INCLUDE_DIR: /usr/include
-- FindGLEW: GLEW_INCLUDE_DIRS: /usr/include
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for SHARED: .so
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for STATIC: .a
-- FindGLEW: GLEW_SHARED_LIBRARY_RELEASE: /usr/lib64/libGLEW.so
-- FindGLEW: GLEW_STATIC_LIBRARY_RELEASE: GLEW_STATIC_LIBRARY_RELEASE-NOTFOUND
-- FindGLEW: GLEW_SHARED_LIBRARY_DEBUG: GLEW_SHARED_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY_DEBUG: GLEW_STATIC_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_SHARED_LIBRARY: /usr/lib64/libGLEW.so
-- FindGLEW: GLEW_STATIC_LIBRARY: GLEW_STATIC_LIBRARY-NOTFOUND
-- FindGLEW: GLEW_LIBRARIES: GLEW_STATIC_LIBRARY-NOTFOUND
-- FindGLEW: GLEW_VERSION_MAJOR: 2
-- FindGLEW: GLEW_VERSION_MINOR: 2
-- FindGLEW: GLEW_VERSION_MICRO: 0
-- FindGLEW: GLEW_VERSION: 2.2.0
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find GLEW (missing: GLEW_LIBRARIES) (found version "2.2.0")
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindGLEW.cmake:213 (find_package_handle_standard_args)
  CMakeLists.txt:492 (find_package)

-- Configuring incomplete, errors occurred!

### Expected results

compile

### Project file & Debug log uploads

$ sudo dnf instlall glew-devel.x86_64 glew-devel.x86_64

$find /usr -name *glew*
/usr/include/GL/eglew.h
/usr/include/GL/glew.h
/usr/include/GL/wglew.h

$ find /usr -name *GLEW*
/usr/lib64/libGLEW.so.2.2
/usr/lib64/libGLEW.so.2.2.0
/usr/lib64/libGLEW.so

$ glewinfo 
GLEW version 2.2.0
Reporting capabilities of display , visual 0x422
Running on a Mesa Intel(R) Graphics (RPL-P) from Intel
OpenGL version 4.6 (Compatibility Profile) Mesa 23.1.9 is supported

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-12-28:
was your building for deps finished successfully?

you can check the directory /home/ats/Downloads/bambu/BambuStudio_dep/usr/local to find whether libGLEW.a exist or not

libglew is built from deps, no need to install in system
we need the static library when building

tmdag, 2023-12-28:
While i didn't get any error, I've realized that destination dependency folder is empty 

```
$ /home/ats/Downloads/bambu/BambuStudio_dep >  tree
.
└── usr
    └── local

3 directories, 0 files

/home/ats/Downloads/bambu/BambuStudio/deps/build >  ll
total 48
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 Blosc
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 Boost
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 boost_polygon
-rw-r--r--. 1 ats ats    29 Dec 27 23:51 boost-user-config.jam
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 Cereal
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 CGAL
-rw-r--r--. 1 ats ats 20888 Dec 27 23:51 CMakeCache.txt
drwxr-xr-x. 1 ats ats  1014 Dec 27 23:51 CMakeFiles
-rw-r--r--. 1 ats ats  1667 Dec 27 23:51 cmake_install.cmake
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 CURL
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_Blosc-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_boost_polygon-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_Boost-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_Cereal-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_CGAL-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_CURL-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_EXPAT-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_FREETYPE-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_GLEW-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_GLFW-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_GMP-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_JPEG-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_MPFR-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_NLopt-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_OCCT-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_OpenCSG-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_OpenEXR-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_OpenSSL-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_OpenVDB-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_PNG-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_Qhull-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_TBB-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_TIFF-prefix
drwxr-xr-x. 1 ats ats    12 Dec 27 23:51 dep_wxWidgets-prefix
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 EXPAT
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 FREETYPE
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 GLEW
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 GLFW
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 GMP
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 JPEG
-rw-r--r--. 1 ats ats 15350 Dec 27 23:51 Makefile
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 MPFR
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 NLopt
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 OCCT
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 OpenCSG
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 OpenEXR
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 OpenSSL
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 OpenVDB
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 PNG
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 Qhull
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 TBB
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 TIFF
drwxr-xr-x. 1 ats ats     0 Dec 27 23:51 wxWidgets

```

```
$ /home/ats/Downloads/bambu >  ls -l
[... 369 lines cut ...]
  OCCT/OCCT.cmake:7 (bambustudio_add_cmake_project)
  CMakeLists.txt:194 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake/Modules/ExternalProject.cmake:3136 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/ExternalProject.cmake:4345 (_ep_add_download_command)
  CMakeLists.txt:80 (ExternalProject_Add)
  FREETYPE/FREETYPE.cmake:13 (bambustudio_add_cmake_project)
  CMakeLists.txt:195 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done (0.8s)
-- Generating done (0.0s)
-- Build files have been written to: /home/ats/Downloads/bambu/BambuStudio/deps/build
```

lanewei120, 2023-12-28:
you need to build it
[image]

go on please

tmdag, 2023-12-28:
oh, damn, sorry for that! my rookie mistake! :)

I appreciate your help!
depts built without issues, Main app had some errors when using -j#

```
[ 12%] Linking CXX static library libmcut.a
[ 12%] Built target mcut
gmake: *** [Makefile:166: all] Error 2
```

removing -jN was of course slower but finished without problems and everything compiled as expected.

Thank you!

closed 2023-12-28 by tmdag