bambulab/BambuStudio #3239: 3FM files keep their spacing like how Prusa sees 3FM files that come from 3D builder

crua9, 2023-12-28:
Coming from Prusa slicer I found a major problem with Bambu slicer. Basically, I normally could build a object mental for multi-color. I could finalize it in 3D builder for windows. Basically I would take each part, subtract from whatever main body, and then re add the parts. This preventing odd bugs. Then save as a 3FM file. 

Normally I would drop the file into Prusa's slicer and it would save the spacing and what not from 3D builder. Where on Bambu slicer it drops the objects to the build plate if they weren't already touching it. 

What should happen is like the Prusa Slicer, the objects should keep their position.

NilSirrah, 2023-12-28:
I know exactly what you mean, I've uploaded a bug explaining this exact issue just now i hope they fix it

crua9, 2023-12-28:
@NilSirrah I just found a work around. It sucks, but at least I can use my 3mf files again. https://youtu.be/7wYbUwuH0cw

bambu123, 2023-12-28:
@crua9 Can you provide a test 3MF file and let's study it?

crua9, 2023-12-28:
@bambu123 You can make one easily with something like 3D builder. I think it still comes preloaded with Windows. If not, you can get it for free from the Windows app store. It shockingly is actually useful for basic task.

Anyways, just throw in a 2 random objects and 1 of them above the other floating. This should be enough to test  If you throw the file into Prusa Slicer the object is as when you saved the file. Without going through back flips, in bambu all the objects are touching the build plate.

crua9, 2023-12-29:
nm 3AM user error.

closed 2023-12-29 by crua9