bambulab/BambuStudio #3240: Modifier coordinate is not matching the coordinate of the main object

NilSirrah, 2023-12-28:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 

### Printer

custom printer profile

### How to reproduce

1.right click main object
2."add modifier"
3.select load

### Actual results

the custom modifier 's CAD coordinate is not matching the main part's CAD coordinate. 
Even if the modifier's CAD coordinate is out of bound from the print bed's dimension the coordinate is still kept.
subsequent adding of modifers will match the previous modifer's CAD coordinate 

### Expected results

The "load" of modifier,negative space, ect... should always match the object's CAD coordinate. 
I am using catia V5 as my cad program.  Even after when the project is saved in the slicer as .3mf file the coordinate should still remembered when opened the next time.
In the example picture i've provided I purposely moved the main body out of the center of the bed. 

Prusa slicer can do this. 

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MklBambu, 2024-01-04:
Can you provide some files for us to debug?

NilSirrah, 2024-01-04:
Yes I have prepared video explaining the issue along with the files

 modifier video.zip
<https://drive.google.com/file/d/1jEoC_fvR89TM0gS-L8BCjS_Mzrs0mNHi/view?usp=drive_web>

T3chC0wb0y, 2024-01-15:
I have the same problem.  Can you provide a timeline for a fix? Trying to move the modifier manually is prone to error.  The coordinates are skewed completely and don't align even when you match them to the coordinates of the main model.

MklBambu, 2024-01-15:
@NilSirrah Sorry，I didn't check the message in a timely manner. I have already watched the video and understood the problem. Can you provide the model file in the video for us to debug? 

Thanks.

NilSirrah, 2024-01-15:
I have attached the files in the previous email in a zip file, but let me
know if u need to send it again

MklBambu, 2024-01-15:
> I have attached the files in the previous email in a zip file, but let me know if u need to send it again
> […](#)

Sorry, I only saw the video in the zip file and did not see the model file.

NilSirrah, 2024-01-15:
There is also one more issue with the modifier, sometimes copying and
mirroring the main body the settings of the modifier will not carry over. I
cannot replicate the issue on demand as it happens kind of randomly

T3chC0wb0y, 2024-01-15:
This coordinate issue also happens for support enhancers.Sent from my iPhone

MklBambu, 2024-01-16:
@NilSirrah I did not find the model file in the previous file. Can you provide the model file here?

zhimin-zeng-bambulab, 2024-02-21:
@NilSirrah  Could you share your model files?