bambulab/BambuStudio #3241: Cant Delete old Files in Models page

Extrusions3D, 2023-12-28:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

windows

### Printer

doesnt matter

### How to reproduce

click on home icon 

### Actual results

there are several projects that i am able to click on and load and even delte from the que. but there are others that we might have moved from their original source folder that i am unable to remove from the projects page no matter how many times i click delete

### Expected results

just need to be able to delete old projects that we have changed the source folder. not even able to update where the source folder was moved to.

[image]

[image]

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-12-29:
Are they have specail names? include space, special charactors?

Extrusions3D, 2023-12-30:
No... I attached pics to show the files

Sent from my Verizon, Samsung Galaxy smartphone

bambu123, 2024-01-08:
@Extrusions3D 
It may not have clicked on the Remove menu.
Can you have a check?

bambu123, 2024-01-25:
@Extrusions3D 
Has this problem been solved? If not, please help provide studio logs.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log