bambulab/BambuStudio #3242: Calibration Issues - Starts calibration print but doesn't update status

lhong820, 2023-12-28:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen Threadripper 3960X
Memory: 64GB
GPU: Nvidia Quadro RTX 4000

### Printer

Bambu P1S

### How to reproduce

1. Calibration Tab
2. Manual Calibration
3. Set parameters
4. Calibrate

### Actual results

Calibration print starts (both flow dynamics and flow rate)
Status doesn't update so values cannot be entered, even after print completes

### Expected results

Status of print updates and allows for next step in calibration to enter in values.

### Project file & Debug log uploads

[3DBenchy.zip](https://github.com/bambulab/BambuStudio/files/13789114/3DBenchy.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/13789134/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Jason-118, 2024-01-02:
Same issue here with P1S latest firmware on both Bambu Slicer and Orca Slicer.

ZachWSays, 2024-01-02:
I've noticed that on the beta firmware it will send the print over internet and then disconnect. It will not reconnect until either power cycling or logging out of the slicer and back in.

KalleFornien, 2024-01-02:
Same issue for me. Updated to new FW today. Doesn't work on multiple devices. Machine and software language EN

crey08, 2024-01-04:
I have/had the same issue. I power cycled my PC and printer. Uninstalled and reinstalled Bambu Studios. Installed and tried Orca and none of that worked. I then downgraded to firmware prior to the one that just came out, recalibrated the vibrations and then I finally got the "next button" to show when doing the calibration for the flow rate! Took hours to get this to work, but I guess it worked for now. Idk if downgrading or a combination of everything together worked or not, but just thought I'd give my notes on my experience.

lhong820, 2024-01-05:
firmware 01.05.01 seems to have fixed this issue. I will complete a calibration and confirm.

crey08, 2024-01-05:
I hope 01.05.01 also fixes the microSD card issue as well...while I was on 01.05.00, I couldn't get any time-lapse videos to load in bambu studio, the models loaded but not the videos which was disappointing, but I downgraded anyway for now since there seemed to be other issues with 01.05.00. Guess we will see once I update again 🤷‍♂️

zhimin-zeng-bambulab, 2024-01-16:
Could you have a try with firmware 01.05.01, it has improved this kind of problem.

zhimin-zeng-bambulab, 2024-02-01:
Does this problem still exist with firmware 01.05.01?

lhong820, 2024-02-01:
This problem has been fixed in 1.05.01.

closed 2024-02-02 by zhimin-zeng-bambulab