bambulab/BambuStudio #3243: Crash on start while logged in

zosman1, 2023-12-28:
### Bambu Studio Version

01.08.02.56

### Where is the application from?

Bambu Lab github releases

### OS version

Linux

### Additional system information

Manjaro Linux, but using ubuntu packages as noamal

### Printer

P1S

### How to reproduce

Install latest build

### Actual results

Program crashes while logged in

### Expected results

Program to not crash

### Project file & Debug log uploads

(bambu-studio:58292): Gtk-CRITICAL **: 16:00:53.253: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
[New Thread 0x7ffda65fd6c0 (LWP 59561)]
[New Thread 0x7ffda75ff6c0 (LWP 59562)]
[New Thread 0x7ffda5dfc6c0 (LWP 59565)]
[New Thread 0x7ffda6dfe6c0 (LWP 59564)]
[Detaching after vfork from child process 59563]
[Thread 0x7ffda5dfc6c0 (LWP 59565) exited]
[Detaching after vfork from child process 59567]
[Thread 0x7ffda6dfe6c0 (LWP 59564) exited]
[New Thread 0x7ffda6dfe6c0 (LWP 59569)]
[New Thread 0x7ffda5dfc6c0 (LWP 59571)]
[Detaching after vfork from child process 59570]
[New Thread 0x7ffda55fb6c0 (LWP 59573)]
[New Thread 0x7ffda4dfa6c0 (LWP 59574)]
[New Thread 0x7ffd8ffff6c0 (LWP 59575)]
[New Thread 0x7ffd8f7fe6c0 (LWP 59576)]
[Thread 0x7ffda5dfc6c0 (LWP 59571) exited]
[Thread 0x7ffd8ffff6c0 (LWP 59575) exited]
[Thread 0x7ffd8f7fe6c0 (LWP 59576) exited]
[Thread 0x7ffda6dfe6c0 (LWP 59569) exited]
[New Thread 0x7ffda6dfe6c0 (LWP 59715)]
[New Thread 0x7ffd8f7fe6c0 (LWP 59716)]
[New Thread 0x7ffd8ffff6c0 (LWP 59717)]
[New Thread 0x7ffda5dfc6c0 (LWP 59718)]
[Thread 0x7ffda5dfc6c0 (LWP 59718) exited]
[New Thread 0x7ffda5dfc6c0 (LWP 59719)]
[Thread 0x7ffda5dfc6c0 (LWP 59719) exited]
[Thread 0x7ffda75ff6c0 (LWP 59562) exited]

Thread 1196 "bambustu_main" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffd8ffff6c0 (LWP 59717)]
0x00007ffff02bb2fe in __GI___libc_free (mem=0x2) at malloc.c:3368
3368      if (chunk_is_mmapped (p))                       /* release mmapped memory. */     

### Checklist of files to include

- [X] Log file
- [ ] Project file

zosman1, 2023-12-28:
I've used the following URL as i've seen a few other people having this issue, but it hasn't resolved the issue for me
`https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage`

dmophir, 2023-12-28:
Confirming this behavior. I have tried multiple versions of Bambu Studio, but none are able to work with the network plugin installed.
### Bambu Studio Version

- BambuStudio_linux_fedora_v01.08.02.56-20231228.AppImage
- BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage
- BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage
- https://github.com/bambulab/BambuStudio/actions/runs/7334871417/artifacts/1135915919

### Where is this application from?
Github releases page, as well as a link in a closed issue to a compiled binary.

### OS Version
Linux

### Additional System Information
Arch Linux, kernel 6.6.8-arch1-1
Dell XPS 9520, specs:
- Intel Core i9 12900HK
- Nvidia RTX 3050 Ti Mobile (4GB)
- 32GB RAM

### Printer
X1 Carbon with AMS

### How to reproduce
Launching the above AppImages via terminal or GUI will result in segmentation fault. This behavior did not occur when `libbambu_networking.so` was removed and `BambuStudio.conf` was edited to not install the network plugin. Reinstalling the plugin will reproduce the error.

### Actual results
As soon as the network plugin was installed, the application crashed when trying to send a sliced object to the printer. Subsequent launches resulted in crashes in seconds of startup, also segmentation faults.

### Expected results
I expected an updated AppImage (which are dated for today) to avoid crashing. This did not occur.

### Project file & Debug log uploads
**Core dump via Systemd-Journal**

```Dec 28 17:50:01 xps9520 gnome-shell[2821]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x2400006
Dec 28 17:50:01 xps9520 dbus-daemon[703]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.7874' (uid=1000 pid=79532 comm="/opt/bambustudio-bin/bin/bambu-studio ")
Dec 28 17:50:01 xps9520 systemd[1]: Starting Hostname Service...
Dec 28 17:50:01 xps9520 dbus-daemon[703]: [system] Successfully activated service 'org.freedesktop.hostname1'
Dec 28 17:50:01 xps9520 systemd[1]: Started Hostname Service.
Dec 28 17:50:01 xps9520 rtkit-daemon[1541]: Supervising 9 threads of 6 processes of 1 users.
Dec 28 17:50:01 xps9520 rtkit-daemon[1541]: Successfully made thread 79594 of process 79577 owned by '1000' RT at priority 5.
Dec 28 17:50:01 xps9520 rtkit-daemon[1541]: Supervising 10 threads of 7 processes of 1 users.
Dec 28 17:50:01 xps9520 rtkit-daemon[1541]: Supervising 10 threads of 7 processes of 1 users.
Dec 28 17:50:01 xps9520 rtkit-daemon[1541]: Successfully made thread 79692 of process 79577 owned by '1000' RT at priority 5.
Dec 28 17:50:01 xps9520 rtkit-daemon[1541]: Supervising 11 threads of 7 processes of 1 users.
Dec 28 17:50:02 xps9520 gnome-shell[2821]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x2400109
Dec 28 17:50:02 xps9520 kernel: bambustu_main[79801]: segfault at fffffffffffffffa ip 00007fc96e6bb2fe sp 00007fc83a1ee0c0 error 5 in libc.so.6[7fc96e644000+15a000] likely on CPU 14 (core 26, socket 0)
Dec 28 17:50:02 xps9520 kernel: Code: 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 f3 0f 1e fa 48 85 ff 0f 84 bb 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d 12 aa 13 00 <48> 8b 47 f8 64 8b 2b a8 02 75 57 48 8b 15 90 a9 13 00 64 48 83 3a
Dec 28 17:50:02 xps9520 systemd[1]: Started Process Core Dump (PID 79804/UID 0).
Dec 28 17:50:07 xps9520 systemd-coredump[79805]: [🡕] Process 79532 (bambustu_main) of user 1000 dumped core.
                                                 
                                                 Stack trace of thread 79801:
                                                 #0  0x00007fc96e6bb2fe free (libc.so.6 + 0x9d2fe)
                                                 #1  0x00007fc960980980 OSSL_ERR_STATE_free (libcrypto.so.3 + 0x180980)
                                                 #2  0x00007fc9609d1c43 n/a (libcrypto.so.3 + 0x1d1c43)
                                                 #3  0x00007fc9609d85b7 n/a (libcrypto.so.3 + 0x1d85b7)
                                                 #4  0x00007fc96e6a7ac1 n/a (libc.so.6 + 0x89ac1)
                                                 #5  0x00007fc96e6aa860 n/a (libc.so.6 + 0x8c860)
                                                 #6  0x00007fc96e72e7cc n/a (libc.so.6 + 0x1107cc)
                                                 
                                                 Stack trace of thread 79532:
                                                 #0  0x00007fc9750e6120 g_source_ref (libglib-2.0.so.0 + 0x56120)
                                                 #1  0x00007fc9750e8ec1 n/a (libglib-2.0.so.0 + 0x58ec1)
                                                 #2  0x00007fc9750eafbb n/a (libglib-2.0.so.0 + 0x5afbb)
                                                 #3  0x00007fc9751481d0 n/a (libglib-2.0.so.0 + 0xb81d0)
                                                 #4  0x00007fc9750eab97 g_main_loop_run (libglib-2.0.so.0 + 0x5ab97)
                                                 #5  0x00007fc9757ed2df gtk_main (libgtk-3.so.0 + 0x1ed2df)
                                                 #6  0x000055fce19e71b5 n/a (bambu-studio + 0x3ac91b5)
[... 48 lines cut ...]
                                                 #6  0x00007fc96e72e7cc n/a (libc.so.6 + 0x1107cc)
                                                 
                                                 Stack trace of thread 79603:
[... cut ...]
Dec 28 17:50:07 xps9520 systemd[1]: systemd-coredump@7-79804-0.service: Deactivated successfully.
Dec 28 17:50:07 xps9520 systemd[1]: systemd-coredump@7-79804-0.service: Consumed 4.064s CPU time.
```

I could not identify any clear error messages from the Bambu Studio logs, uploading a complete log.
[debug_Thu_Dec_28_17_50_01_79532.log](https://github.com/bambulab/BambuStudio/files/13790061/debug_Thu_Dec_28_17_50_01_79532.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

lanewei120, 2023-12-29:
@zosman1 @dmophir 

was the previous app image working before?
if yes, what is the version of that

zosman1, 2023-12-29:
Public beta 3 works for me
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.01.57
specifically i'm using the ubuntu release

dmophir, 2023-12-29:
`BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage` was working for me until the network plugin update was pushed. That happened yesterday on my machine.

lanewei120, 2023-12-29:
[linux_libraries.zip](https://github.com/bambulab/BambuStudio/files/13790484/linux_libraries.zip)
@zosman1 @dmophir 
please try these libraries, extract them into ~/.config/BambuStudio/plugins/
and don't upgrade the networking plugins currently

lanewei120, 2023-12-29:
we have also released a new network plugins just now
after using the above plugins to avoid crash, you can upgrade to the newest plugins, and the login should work now

castlevania1983, 2023-12-29:
> we have also released a new network plugins just now after using the above plugins to avoid crash, you can upgrade to the newest plugins, and the login should work now

This worked for me on Arch, thank you!!

dmophir, 2023-12-29:
Confirmed, using the plugins in the provided zip file solved my issue. I was able to open a project, add a new step file, slice it, and send it to my X1 Carbon via WiFi. Thank you!

zosman1, 2024-01-01:
This fixes the issue for me! @lanewei120 Would you like me to add documentation anywhere to point users to this fix? If not i'll go ahead and close this issue.

Thank you so much for the speedy fix!

lanewei120, 2024-01-01:
@zosman1 we have also fixed it in the newest plugin,
 if downloading the plugin directly, this issue has also been fixed

no need to write documentation anymore

thanks

closed 2024-01-05 by zosman1