bambulab/BambuStudio #3244: Cut 'snap' connector is not working correctly

Zulex, 2023-12-28:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Add object
2. Cut object
3. Add 'snap' connector to object
4. View object, on the 'hole' side you should see a circle carved out with a line masked (which makes sure the snap can clamp)
5. After confirming cut, this line which is masked, is not there anymore and the whole circle is carved out. 

### Actual results

[image]
[image]

### Expected results

I was expecting to see the masked line also in the final cut. This makes that the snap isn't working correctly to my understanding. 

### Project file & Debug log uploads

[Test.zip](https://github.com/bambulab/BambuStudio/files/13789937/Test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye01, 2023-12-29:
Actually, after printing it out, the snap on my end can work.Can you try it?

Haidiye01, 2023-12-29:
this is the rendering effect is not the final model
[image]
this is the final cut model
[image]

Zulex, 2023-12-29:
The final model does snap a little, but not as much as expected. However, I am not sure what to expect ofcourse. 

But I was wondering if that line over de circle is just for indication, or that it should be printed that way? As it is definitely not printed that way.

Haidiye01, 2024-01-09:
[image]
Finally, print the 3D model according to this

closed 2024-01-16 by Haidiye00