bambulab/BambuStudio #3246: [BambuLab X1C+AMS Firmware] Broken Filament above Print Head Extruder handling problem.

cederom, 2023-12-29:
### Bambu Studio Version

1.8.2

### Where is the application from?

Bambu Lab github releases

### OS version

FreeBSD 13.2-RELEASE AMD64 / Linux VM

### Printer

BambuLab X1C+AMS

### How to reproduce

1. I am using AMS with BambuLab X1C.
2. The filament got somehow broken just above the the print head.
3. AMS went crazy and the whole printer had to be restarted with power switch several times until I figured out what was the cause.
4. I could not replace the filament in the AMS on the printer panel.
5. I got error screen that "Failed to pull out the filament from the extruder. (..) [0700 8003 153109]". There was only "Retry" and "Close" button. Retry did not solve the problem. Close seemed to put printer in an undefined state.
6. After some time I have noticed that the filament was broken above the print head (almost impossible to notice without careful examination).
7. I have used manual printer control panel to set print head temperature to 230C and then manually extruded the broken piece of material. That solved my problem.

### Actual results

1. Printer and AMS goes crazy on filament removal.
2. There is no error dialog for handling the problem from a printer panel.

### Expected results

1. If the error message says that filament may be broken inside extruder then there should be additional step to remove it with a single click (i.e. additional Purge Extruder) and maybe separate error handling dialog.

2. Pull back filament operation should make sure that if the filament slot is empty at AMS then also extruder must be empty, if not then broken filament in the extruder happens and could be handled with a dedicated dialog (point 1).

3. The proper way to purge broken filament from the extruder is to manually control printer, set head temperature, retract broken filament until new one can be loaded fitting directly the print head extruder.

### Project file & Debug log uploads

Photos of the problem are attached below:

1. Standard filament replacement dialog:
[image]

2. Filament pull out error. No check if filament is still in the extruder. No option to handle this error easily.
[image]

3. Cause of problem: broken filament above extruder.
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-01-02:
it seems a printer related issue,
it is better to create a customer ticket https://bambulab.com/en/my/support/tickets/create here
which will has more faster response

anyway, I will forward it to our printer colleague also

cederom, 2024-01-02:
Thank you @lanewei120 :-) Its more a firmware feature idea.. should I report those here or another github repo or the customer service? :-)

lanewei120, 2024-01-02:
> Thank you @lanewei120 :-) Its more a firmware feature idea.. should I report those here or another github repo or the customer service? :-)

yes, you'd better report to the customer service https://bambulab.com/en/my/support/tickets/create

cederom, 2024-01-02:
Thank you @lanewei120 :-) All The Best in New Year 2024!! :-)

closed 2024-01-02 by cederom

lanewei120, 2024-01-02:
> Thank you @lanewei120 :-) All The Best in New Year 2024!! :-)

Thank you @cederom 

Happy New Year~