bambulab/BambuStudio #3247: When "A G-Code path goes beyond plate boundaries" error occurs, inform the user where it is occurring.

elpeterson, 2023-12-29:
**Is your feature request related to a problem? Please describe.**
A G-Code path goes beyond plate boundaries - this error occurs during slicing of multiple identical objects (in my scenario), often times the supports are not trimmed at the edges of the build plate like what happens when a single object runs close to the edge of a build plate. When slicing several, closely positioned, objects this error is thrown but does not inform the user where the error is occurring.

**Describe the solution you'd like**
The same way Bambu Studio highlights where collisions are happening with two models, highlight where a model is going beyond plate boundaries

**Describe alternatives you've considered**
Highlight the edge of the build plate where it's occurring if finding the exact area of the object is too difficult

**Additional context**
In this screenshot, both edges of the object are an issue. If I had 25 of these on the build plate, but only one was causing an issue, knowing which object or which edge would be incredibly helpful instead of just having a single, vague "A G-code path goes beyond plate boundaries" error.
[image]

Bald-e, 2024-04-11:
Pretty please? This is driving me nuts

gmiller123456, 2024-04-26:
Same here, no idea what it's complaining about.  I've already deleted quite a few.  Since the slicing takes a couple of minutes for this, it's pretty frustrating just guessing and re-trying.
[image]

IamGimli, 2024-07-05:
Would love to see this feature added as well. I have a model that won't slice and I have no idea where the issue is, all of the measurements are within limits.

k1n6b0b, 2024-12-24:
Yup, been staring at my plate for 10min and still cant find it :( Would be helpful to have this pointed out