bambulab/BambuStudio #3248: Printer gone everytime

sajungzak, 2023-12-29:
### Bambu Studio Version

most recent one

### Where is the application from?

Bambu Lab Official website

### OS version

11

### Printer

p1s

### How to reproduce

open the app

### Actual results

printer is gone everytime

### Expected results

it should save it like it did yesterday

### Project file & Debug log uploads

ffffffff

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-12-29:
Hi. @sajungzak 
Are you saying that the printer is no longer in your printer list.

SaltWei, 2024-01-18:
Hi. @sajungzak
Are you saying that the printer is no longer in the printer list of device page of bambu studio?

walterwongbbl, 2024-03-02:
This issue will be closed. Thank you very much for your feedback.

closed 2024-03-02 by walterwongbbl