bambulab/BambuStudio #3249: Loss of connectivity with printer

SemperFi89, 2023-12-29:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 23H2 x64 Build 22631.2861

### Additional system information

Intel Core i7-10700K, Nvidia GeForce RTX 3050

### Printer

X1C with 4 AMS's

### How to reproduce

Open Bambu Studio
Goto Device
Stay there

### Actual results

Bambu Studio will lose connection with the printer after a short period.  After 10-30 seconds, it will reconnect, then lose connection again.  You do not have to have any project open.  Also, when Bambu Studio does reconnect, you will hear one of the AMS cycles its filament in slot 1.  If you leave the Device tab, the AMS cycling will stop.  This started after I upgraded Bambu Studio to 1.8.2.56
[image]
lose

### Expected results

Bambu Studio should not keep loosing connection.

### Project file & Debug log uploads

[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/13792158/BambuStudio.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Bigguy34, 2023-12-30:
I have the same issue, it started happening after I upgraded the network driver for bambu studio. It is preventing uploads to the printer via LAN only mode, specifically for larger files.

GovGeek, 2023-12-31:
As discussed here https://www.reddit.com/r/BambuLab/comments/18ogdc6/comment/kfmj9ub/?context=3 , I am also seeing the same issue. Uninstalling and reverting back to v01.07.07.89 restores connectivity as expected.

GovGeek, 2023-12-31:
> As discussed here https://www.reddit.com/r/BambuLab/comments/18ogdc6/comment/kfmj9ub/?context=3 , I am also seeing the same issue. Uninstalling and reverting back to v01.07.07.89 restores connectivity as expected.

Disregard, per this article https://github.com/bambulab/BambuStudio/issues/3122 syncing the clock fixed it right up!

walterwongbbl, 2024-01-02:
Hi @Bigguy34 
please take a look at this.  https://github.com/bambulab/BambuStudio/issues/3122

closed 2024-01-18 by walterwongbbl