bambulab/BambuStudio #3250: Filament presets not working

MathiasWeixlbaumer, 2023-12-29:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Microsoft Windows 11 Home; 10.0.22631

### Additional system information

CPU: intel i5-13600KF
GPU: Nvidia Geforce RTX3080
RAM: 2x16GB DDR5

### Printer

BambuLab X1C+AMS

### How to reproduce

Click on the settings gear for "Set filaments for use". Every time, i want to deselect some system filament presets (like overture) I click on the confirm-button, but the changes do not take action.

Next problem - when creating custom filament presets, everything works fine. But after I created one, I want to set the preset filament in the AMS. This works for a while and after a few seconds (or another custom preset is set to another AMS slot) the selection disappears and a "?" is displayed in the edited slot/s.

### Actual results

[image]
I deselected the Overture preset and clicked confirm.

[image]
Then, I opened the menu again and Overture is selected again.

**Next problem:**
[image]
I created a custom filament preset for the brand eSun and set it to an AMS-Slot:
[image]

Then, after  a few seconds of waiting and doing nothing, this happens every time:
[image]

### Expected results

Quite obvious, the selected system filament presets should take action and the custom filament preset should stay when set to an AMS-Slot.

### Project file & Debug log uploads

This issues are actually not dependent on a single project.
[Körper2.zip](https://github.com/bambulab/BambuStudio/files/13792304/Korper2.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/13792314/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DanielHoyland, 2023-12-29:
Hey @MathiasWeixlbaumer,

I had this issue myself and it seems to be due to the AMS.

If you set the filament specifically on the printer and then try to change it within Bambu Studio there seems to be some sort of conflict and the printer takes precedence.

Try setting the filament directly on the printer first and then changing it in Bambu Studio,

I hope this helps.

Sterbo88, 2023-12-29:
I also have the same problem... the AMS does not accept costum filaments in the presets... and it is not possible to select it from the printer

east005, 2023-12-30:
I also have the same problem even some custom filaments do not show up in the settings menu but in filament selection drop down menu which is really confusing

DanBao-Bambu, 2024-01-03:
@MathiasWeixlbaumer  @east005  @Sterbo88 
For this question:
> This works for a while and after a few seconds (or another custom preset is set to another AMS slot) the selection disappears and a "?" is displayed in the edited slot/s.

Indeed, the current collaboration between Studio and printer may confuse you, here is the solution:
Firstly, confirm your method of creating custom Filament. There is an issue with one workflow, which is to select "Copy Current Filament preset" and then select a user preset under a certain printer. This will result in the created Filament being unable to be transferred to the printer. If this is your creation method, please change the creation method, and this issue will be fixed in the next version.
Secondly, if you have correctly created a custom Filament, please go to the screen of the printer and enter the Filament settings interface, because only when the printer enters this interface will the printer go to the cloud to retrieve data. After obtaining the data from the printer, you can select AMS in Studio.

MathiasWeixlbaumer, 2024-01-03:
> @MathiasWeixlbaumer @east005 @Sterbo88 For this question:
>
>
> Indeed, the current collaboration between Studio and printer may confuse you, here is the solution: Firstly, confirm your method of [...]

Ahh okay thanks! Thats exactly the way I created all my custom presets. I just want the existing brand of the filament with the custom set print settings. Is there also gonna be a way to select the self-created brands for an AMS-slot? I created a custom brand (3DJake), but when setting a new filament to a slot, the brand is not displayed.
Is there also gonna be a fix for the system filament presets bug (assuming I haven't made a mistake)?

Sterbo88, 2024-01-03:
> @MathiasWeixlbaumer @east005 @Sterbo88 For this question:
>
>
> Indeed, the current collaboration between Studio and printer may confuse you, here is the solution: Firstly, confirm your method of [...]

I honestly didn't understand the solution.... but whatever I do, it doesn't work… I create custom filament, select sunlu from the drop-down menu... I copy current filament preset, but then from the x1c it doesn't let me select this filament and if I insert it on the ams from bambu studio it gives me the same problem

east005, 2024-01-03:
@DanBao-Bambu 

> Firstly, confirm your method of creating custom Filament. There is an issue with one workflow, which is to select "Copy Current Filament preset" and then select a user preset under a certain [...]

I can confirm that i created the custom filament as you mentioned with the known bug.
I deleted all  Filaments i could in the custom settings then I created new custom filaments in the correct way as you mentioned
Like this:
[image]

> Secondly, if you have correctly created a custom Filament, please go to the screen of the printer and enter the Filament settings interface, because only when the printer enters this interface will [...]

If i do the second steps as you mentioned everything works fine.....
thanks a lot for the clarification

the only issue which remains is that i can not delete all custom created filaments

the new created filament list looks like this:
[image]

but when i look in the filament dropdown there are a lot of custom filaments which i can not delete (yellow marked).
[image]

@DanBao-Bambu do you also have a workaround for that issue?

Sterbo88, 2024-01-03:
> @DanBao-Bambu
> 
> > Innanzitutto, conferma il tuo metodo di creazione del filamento personalizzato. Si è verificato un problema con un flusso di lavoro, ovvero selezionare "Copia preimpostazione filamento corrente" e quindi selezionare una preimpostazione utente sotto una determinata stampante. Ciò comporterà che il filamento creato non potrà essere trasferito alla stampante. Se questo è il tuo metodo di creazione, modifica il metodo di creazione e questo problema verrà risolto nella prossima versione.
> 
> Posso confermare di aver creato il filamento personalizzato come hai menzionato con il bug noto. Ho eliminato tutti i filamenti che potevo nelle impostazioni personalizzate, quindi ho creato nuovi filamenti personalizzati nel modo corretto come hai detto in questo modo: ![Immagine](https://private-user-images.githubusercontent.com/38375399/294004150-87433213-b6e6-4933-8c02-f5773a49dda8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzMTAxMDQsIm5iZiI6MTcwNDMwOTgwNCwicGF0aCI6Ii8zODM3NTM5OS8yOTQwMDQxNTAtODc0MzMyMTMtYjZlNi00OTMzLThjMDItZjU3NzNhNDlkZGE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAzVDE5MjMyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0NzA3OWNhN2VmZDY2Y2UxYjhlNWQ4MDdkYmY2MWM2NjViZWRhMzQwODJmZTFiZGM2YTQzNDAxOWIwYWZmZjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Oo5xtbv_gfuEmIj64gQ1VNVjByRs6-SK6q-yJnf-6I4)
> 
> > In secondo luogo, se hai creato correttamente un filamento personalizzato, vai sullo schermo della stampante e accedi all'interfaccia delle impostazioni del filamento, perché solo quando la stampante accede a questa interfaccia la stampante andrà nel cloud per recuperare i dati. Dopo aver ottenuto i dati dalla stampante, è possibile selezionare AMS in Studio.
> 
> Se eseguo i secondi passaggi come hai detto, tutto funziona bene..... grazie mille per il chiarimento
> 
> l'unico problema che rimane è che non riesco a eliminare tutti i filamenti creati su misura
> 
> il nuovo elenco di filamenti creato assomiglia a questo: ![Immagine](https://private-user-images.githubusercontent.com/38375399/294005657-aa2493fd-dd6f-4caf-8883-c53f63b92e20.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzMTAxMDQsIm5iZiI6MTcwNDMwOTgwNCwicGF0aCI6Ii8zODM3NTM5OS8yOTQwMDU2NTctYWEyNDkzZmQtZGQ2Zi00Y2FmLTg4ODMtYzUzZjYzYjkyZTIwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAzVDE5MjMyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUxOWNiYWRlYmM3ZmY4YWQyNjg0Y2ZlMWFkYmQwNGFmYjUwNGE4Y2I4YTU0YzUzOGNjNzFhMmJhMGE5ZDRiNWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.mKcG1jqGuIBDHwuTyl3KmWizCQUm1ztGYvf1g4wwZBM)
> 
> ma quando guardo il menu a discesa dei filamenti ci sono molti filamenti personalizzati che non riesco a eliminare (contrassegnati in giallo). ![Immagine](https://private-user-images.githubusercontent.com/38375399/294008205-764644d9-da85-40d8-a643-ff8b4a713246.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzMTAxMDQsIm5iZiI6MTcwNDMwOTgwNCwicGF0aCI6Ii8zODM3NTM5OS8yOTQwMDgyMDUtNzY0NjQ0ZDktZGE4NS00MGQ4LWE2NDMtZmY4YjRhNzEzMjQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAzVDE5MjMyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY2ODMxNDk2ZjIzMGM5MzgxMDUyNzRhZTkxZTBlYjMyNGMxYzQ5MmY5NmQ3YWZiNjY1OTcyNGJmNTkwZTBiMWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8j_Yx8cglel-YJnNtu5MeblLQfOiAEk9tvhWK4iU73g)
> 
> @DanBao-Bambuhai anche una soluzione alternativa per quel problema?

This method didn't work for me... after a factory reset it works...

bam-winterfell, 2024-01-05:
hi@east005
You can delete these presets by performing the following actions
1.On the prepare page, set the material as the preset material to be deleted
2. Click on the material editing button
3. Click the "X" button V
[image]

Naninani, 2024-01-05:
This is very buggy to say the least.
When a preset is created and assigned to the AMS, the AMS does not record it and then it switches back to Generic. Thus, if one chooses a slot it tries to put it to another in the ams. The sync is completely busted.

DanBao-Bambu, 2024-01-06:
[#3087](https://github.com/bambulab/BambuStudio/issues/3087)

DanBao-Bambu, 2024-01-06:
@Sterbo88 
> I honestly didn't understand the solution.... but whatever I do, it doesn't work… I create custom filament, select sunlu from the drop-down menu... I copy current filament preset, but then from the [...]

This question should be about the Filament you selected when creating it. Please select the Filament provided by the system, such as "Bambu PLA Basic" or "General PLA". If the Filament you have chosen is a custom Filament, I am not sure if this Filament has been damaged, which may cause problems with your newly created Filament as well. The next version will prompt you in Edit Filament which presets need to be deleted. When you delete all problematic presets, I believe all the workflows in Studio are available. At that time, you can create custom Filaments in any way you want to use.

DanBao-Bambu, 2024-01-06:
@east005 I know your confusion, these exist in the optional list, but the Filaments that cannot be displayed in the Custom Filament list can be considered as broken Filaments. They are problematic and cannot be found by the printer (cannot be displayed in the printer).

Here are the solutions:
At present, you can manually delete one by one.
[image]

Another method is to wait for the next version, the new Studio version will make these problematic Filaments reappear in the Custom Filaments list, and will remind you of which items to delete in the problematic items. At this time, your problematic presets will appear in one item, and you can delete them all at once.

I'm sorry for the inconvenience caused to you.

Naninani, 2024-01-06:
> @east005 I know your confusion, these exist in the optional list, but the Filaments that cannot be displayed in the Custom Filament list can be considered as broken Filaments. They are problematic [...]

I deleted all my presets and created new ones each following what you suggested. When I assign a preset to the AMS in the device tab and get back to the filament assignment in the device tab it gets reset to Generic PETG
[image]
and the ams shows empty filament
[image]

east005, 2024-01-06:
@bam-winterfell and @DanBao-Bambu 

> hi@east005
> You can delete these presets by performing the following actions
> 1.On the prepare page, set the material as the preset material to be deleted
> 2. Click on the material editing button
> 3. Click the "X" button
> image

Thanks for the hint how to delete the filaments worked perfektly for me

designgears, 2024-01-18:
> @MathiasWeixlbaumer @east005 @Sterbo88 For this question:
>
>
> Indeed, the current collaboration between Studio and printer may confuse you, here is the solution: Firstly, confirm your method of [...]

Did the fix not make it into the release?

DanBao-Bambu, 2024-01-22:
@designgears The method of creating a Filament has been fixed, and now no matter how you create a Filament, there will be no problems.
However, there are still some prerequisites when using it, such as enabling cloud synchronization and accessing the Filament settings interface from the printer when creating the Filament for the first time to obtain data.
We will solve this problem in the future.

Gil80, 2024-03-22:
I'm still not able to create filament presets with the latest Studio as of 22/03/2024.
[image]

No matter which method of creating a custom filament, I keep running into issues with creating filament presets.

Is there any solution to this?

DanBao-Bambu, 2024-03-24:
@Gil80 I'm not sure if I understand your doubts correctly. Here are some of my explanations and solutions.
The relationship between Filament and Filament Preset is as follows: a Filament can have a Preset for that Filament under a single model.
The operation in your picture is to add a new model preset to the current Filament.
In the picture, you already have a preset for the X1C 0.4 nozzle. When adding the preset, you still selected the X1C 0.4 nozzle model, so it will prompt that the preset already exists.
If you want to create a new Filament, please click on New Filament in the Custom Filament section of the Filament wizard.

Gil80, 2024-03-24:
@DanBao-Bambu I think this whole flow is quite confusing.

As I understood this, I had to create a so-called "Template" - example: **eSun ABS Plus**
Then under this template, I can add my existing profiles that have their flow and pressure advanced calibrated so I can see it also in the printer's LCD screen.

I simply wanted to add missing filament brands or missing filament types (plus, carbon,etc) so I can also select them in the LCD screen of the printer with their respective pressure advance values.