bambulab/BambuStudio #3251: Blank page when opening BambuStudio

PierreDeQuebec, 2023-12-29:
### Bambu Studio Version

1.8.2.6 (BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage)

### Where is the application from?

Bambu Lab github releases

### OS version

Kubuntu 23.04

### Additional system information

4 x Intel Core i5-7400 @ 3 GHz
15,6 GB RAM
NVIDIA GeForce GTX 1660/PCIe/SSE2
Linux 6.2.0-39-generic (64 bit)
X11

### Printer

Bambu Lab P1P

### How to reproduce

1. Download Image BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage
2. Démarrer l'application.

### Actual results

Here are two screenshots. The first shows the blank page as it appears after starting the application. The second shows, as an example, the "Prepare" tab where everything seems normal.
[image]
[image]

### Expected results

See the home page to log in to my Bambu Lab account.

### Project file & Debug log uploads

None.

### Checklist of files to include

- [X] Log file
- [X] Project file

PierreDeQuebec, 2023-12-30:
I found a solution after digging through the issues. In particular, the message [SnickCH](https://github.com/bambulab/BambuStudio/issues/2567#issuecomment-1770499986) directed me to the solution applicable under KDE. I added the environment variable in my launcher definition 'BambuSt
udio.desktop' (see image).

[image]

bambu123, 2024-01-02:
> I found a solution after digging through the issues. In particular, the message [SnickCH](https://github.com/bambulab/BambuStudio/issues/2567#issuecomment-1770499986) directed me to the solution applicable under KDE. I added the environment variable in my launcher definition 'BambuSt udio.desktop' (see image).
> 
> ![Bambu desktop solution under Linux](https://private-user-images.githubusercontent.com/13820713/293481041-6c89b306-fd78-4b2a-8d1a-b16416a9e1dc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxNjA2NzUsIm5iZiI6MTcwNDE2MDM3NSwicGF0aCI6Ii8xMzgyMDcxMy8yOTM0ODEwNDEtNmM4OWIzMDYtZmQ3OC00YjJhLThkMWEtYjE2NDE2YTllMWRjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAyVDAxNTI1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEwOWVhOWNmZWVlNGJmYjFmNTkyY2VmODgwNGJjZDI3NzIzMDU2NmM2NGE0M2QwMjUzMTlmMTJkOWQ1ZmYxYTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QmxJSBDZg7G2ttStoSJo9n_Ko50TJH8kWrPrf2FjZGQ)

Yes, that's it.