bambulab/BambuStudio #3252: Customising Filaments Isn't User Friendly

daniel77261, 2023-12-29:
Customising your choice of filaments in Bambu Studio isn’t straightforward or user-friendly. It’s cumbersome, counter-productive and a chore.

Unable to Create a new Filament from Scratch

There is no option to create a new filament from scratch, instead, you must copy from an existing filament. This means it’s very easy to accidentally copy over settings you don’t want/need and not have a desired configuration.

[image]

Unable to Specify Type or Vendor

This has been moderately solved in v1.8.2 but it’s still unintuitive. When editing or copying a filament, you should be able to change the ‘Type’ or ‘Vendor’ at any stage - like all the other fields. But instead, it’s greyed out.

[image]

Custom Filaments - an Organisational Nightmare

The introduction of custom filaments, whilst appreciated, is an overly cumbersome way of fixing my previous complaint where the vendor field should really just be plain text.

Sadly, it’s not been properly thought through and creates more issues than it fixes.

Type field: Unlike the ‘Vendor’ field, you cannot specify a filament type that isn’t in the list. That means that filaments such as ePLA, PLA+ or PLA Blend are not available and instead have to be entered into the ‘Serial’ field.

[image]

Serial field: This field should not be mandatory. Not every brand uses the same naming convention as Bambu Lab and so you’re forced to create filaments with silly names such as ‘Prusament PLA PLA’.

[image]

Filament preset: This goes back to my first point where you’re forced to copy another filament at every stage. You cannot create a new filament without copying the preset of another one which again leaves room for error. Furthermore, if a filament type doesn’t exist you have to enable a system filament to be able to copy from it.

[image]

Custom filament naming conventions: The fact that you cannot edit the name of the custom filament is a huge oversight. The options you pick for the ‘Vendor’, ‘Type’ and ‘Serial’ create the name which is then trailed with ‘@Bambu Lab X1 Carbon 0.4 Nozzle’. This is hideous. Any attempt to change the name creates a new filament which makes the aforementioned one redundant. Leading to my next issue.

[image]

Unable to delete a filament which has been copied from a custom filament preset: If you copy a custom filament, the filament you create is nested under the filament you copied it from (even though the tree diagram doesn’t show this). As a result, if you try to delete the filament you copied from your newly created filament will also be deleted.
image

[image]

AMS Filament, Project-inside presets, User presets and System presets

This one speaks for itself, there are far too many options for filament types, leading to a really clunky UI.

[image]

Would love to see this interface improved to make everything a bit more fluid and user-friendly.

Sonicboom247, 2023-12-31:
Going to have to agree here, it is quite cumbersome and overly counterintuitive. I understand the product is fairly new and needs to get to market so I will learn to use it for now, but it really needs the aforementioned changes or something similar..

hellium, 2024-01-01:
Agreed, im playing with the feature for the last hour and my brain hurts, it just isnt intuitive in any way, no matter from what angle i look at it it just feels wrong.

Misiu, 2024-02-12:
Ideally, there should be a separate window to manage custom filament presets (and share them with the community!)
Also, is there a procedure for adding filament preset to the official set (that is in the app)?
There are many great filament companies, but BambuStudio is missing a lot of them, we can add them ourselves, but I think all of us will benefit from a larger set of tested and approved filaments.

ideaalab, 2024-02-15:
A separate window for managing filaments would be really nice. And maybe the MakerWorld page could host filament settings, so for same printer and same filament you could download a tested setting without having to do all the calibration test yourself.
But number one priority is to be able to rename filaments I think.

daniel77261, 2024-05-16:
I'd love to have an official response from Bambu Lab on the issues posted here, of which I have a new one.

We received our second X1-C today and I’ve noticed a new issue with customising filaments.

Any Flow Dynamics calibrations you have saved are saved to the printer. That means for every printer you setup, you have to re-create the Flow Dynamics calibrations from your previous printer.

Is this expected behaviour? With the same printer and material surely the results should be the same.

armishmajerj, 2024-08-31:
Another fun issue with adding/naming custom filaments is if you do an auto flow rate calibration and save the profile it allows you to keep the name without limits. But when I tried a manual calibration and tried using the same naming scheme I reached a character limit

MillyRamsey, 2024-11-26:
Not being able to rename filament profiles something that is very frustrating for me, too. I want to have my filament list nice and neat, and as it is (Studio creating my new filament profiles with pre-generated names, then me trying to rename them), it's a mess!