bambulab/BambuStudio #3253: Total layers not shown in device screen

ercdvs, 2023-12-29:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

P1S

### How to reproduce

1. Start a print
2.  Go to 'Device' screen
3.  Observe layers printed / remaining and see that total layers is '0'

[image]

This has persisted across every print and even closing BS and restarting

### Actual results

[image]

Total layers are 0, but current layers are counted

### Expected results

I expect to see total layers for the print

### Project file & Debug log uploads

[bugreport.zip](https://github.com/bambulab/BambuStudio/files/13795796/bugreport.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2024-01-18:
Could you please provided the gcode you printed. We need to check whether its slicer issue or printer issue. The total layer number is written into gcode file by slicer, but in device page, it is from printer by reading gcode.

TCC-up, 2024-01-18:
Hello, we need a joint analysis of the logs of the equipment, can you help to submit a work order and give us the corresponding work order number, including the logs of the corresponding time period, thank you for your help!