bambulab/BambuStudio #3254: Add option to disble "fill gaps" to further improve print quality and print speed for specific models

mrcg, 2023-12-29:
**Is your feature request related to a problem? Please describe.**
**In original Prusa Slicer i have often disabled the "fill gaps" option to improve print speed and print quality for some particula objects for which lot of gap fill tasks have been created. The improvement in time and quality can be massive depending on the printed object.
Unfortunately it's not possible to disable the "fill gaps" feature in Bambu Studio.**

**Also lot of other users have requested to have the option to disable the "fill gaps" in Bambu Studio. See link.
https://forum.bambulab.com/t/allow-disabling-gap-filling/1510/7**

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
**The option to enable/disalbe the "gap infill" feature should be made user configurable.
Same is it is also available in official Prusa Slider.** 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
**No alternative possible.**

**Additional context**
Add any other context or screenshots about the feature request here.
Screenshot from Prusa Slicer attached.
[image]

man55, 2024-01-01:
This is a useful feature and is sorely missed

NVNDO, 2024-05-16:
You have my vote!