bambulab/BambuStudio #3255: Models won't render in the Prepare or Preview pane on Mac

Broctune, 2023-12-30:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Ventura 13.4.1 (c)

### Additional system information

CPU: 2.3 GHz 8-Core Intel Core i9
Graphics: AMD Radeon Pro 5500M 4 GB
  Intel UHD Graphics 630 1536 MB
Memory: 16 GB 2667 MHz DDR4

### Printer

X1 Carbon

### How to reproduce

I'm simply trying to prepare a model in Bamboo Studio on my Mac. I've tried to open files directly from makerworld.com as well as directly open 3MF files from my local file system. Bamboo Studio appears to open the file, but there is no rendering in the Prepare tab. I can adjust attributes like changing the printer, filament, etc. In the Process section of attributes, I can switch from Global to Objects and see the model listed under Plate 1. The model view is simply blank.

If I click Slice I'm taken to the Preview tab but there is no rendering there, either, and it doesn't appear that it's actually sliced the model. There is no delay or model stats, etc.

I've tried to create new projects and import STL and other files but get the same result - no rendering of the object, plate, etc. Just a blank view port. I never see any errors or anything that would indicate something is wrong.

I'm logged in and can connect to the printer. I can view the camera from the Device tab and see which filaments I have loaded.

I've tried uninstalling, rebooting, reinstalling, re-downloading but get the same result. 

I switched back to a Windows machine and am able to print as usual using the save version. 

### Actual results

[image]

### Expected results

I expect to open the Project / Model and see the build plate with the objects available for preparation, slicing and printing. 

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/13797013/Archive.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Killdozer666, 2023-12-31:
I have the same issue on Bambu Studio 1.8.2.56 for Linux

Haidiye01, 2024-01-02:
is there any logs here /Users/user.name/Library/Application Support/BambuStudio/log?

you can find the guide
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
to export logs
thanks

This situation is highly likely to be an opengl driver issue

Killdozer666, 2024-01-13:
I had the same issue on Linux mint, and disabling webkit by an environment variable fixed it for me. See this issue for ideas.
[https://github.com/SoftFever/OrcaSlicer/issues/2703#issuecomment-1890293931](url)
Its been a while since I used macOS, but editing your application shortcut with a similar command, might help.  Start there.

matt3948, 2024-12-04:
There is a workaround for this issue (Applies to Prusa Slicer, Orca, and pretty much any others that have this openGL driver issue). The workaround is to set: _Force Accelerated Compositing mode to be always off_ with an environmental variable:  

`WEBKIT_DISABLE_COMPOSITING_MODE=1 `

On mac you can use Terminal to run the below command which will apply for the session:
_(I'm not sure how to apply to a specific application / add it to a shortcut)_
`export WEBKIT_DISABLE_COMPOSITING_MODE=1`

For a shortcut example (Linux) you can use a command like so:

`Exec=sh -c "env WEBKIT_DISABLE_COMPOSITING_MODE=1 __GLX_VENDOR_LIBRARY_NAME=mesa  __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json /home/wingedhussar/Documents/Applications/Bambu_Studio_linux_ubuntu_20.04_v01.10.01.50_715820e9285ba39313ac291604726ac9.AppImage %F"`

Haidiye01, 2024-12-04:
ok,i have received your letter ,thank you!

PeterGrace, 2024-12-06:
I came to the issues to report this rendering issue for the Linux AppImage, but @matt3948 's resolution worked for me in an ubuntu 24.04 variant.

ahoneybun, 2025-01-03:
I see this in the Ubuntu 24.04 AppImage but the one for Fedora (on Bazzite) seems to work fine?

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

BookWyrm114, 2025-04-08:
I'm experiencing this issue as well on the latest flathub release

[image]