bambulab/BambuStudio #3257: On a <=1170px wide display, the "Print plate" button is inaccessible

maruel, 2023-12-30:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04 (issue is not OS specific)

### Printer

Bambu Lab X1C

### How to reproduce

1. Have a monitor less than 1200px wide
2. Try to print anything

### Actual results

The "Print plate" button is impossible to access and there is no scroll bar to access it.
[image]

My monitor is narrow so this is a significant problem.

For power users, it can be sometimes useful to work split screen, CAD/MakerWorld on one half and slicer on the other half. Even on a standard 1920px wide monitor, Bambu Studio cannot be used with half a screen since the "Print plate" button is inaccessible.

### Expected results

Create a menu item in the top left menus to access "Print plate" would be a reasonable workaround.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [X] Project file

maruel, 2023-12-30:
Note that this is different from issues #1304 and #2891 where there IS a workaround: the horizontal bar can be used to access the printer commands in the Device page. The issue described here has no workaround as far as I know.

walterwongbbl, 2024-01-18:
Our horizontal scrollbar does not seem to be working, and that is indeed an issue. Does the horizontal scrollbar appear when you resize the window?

maruel, 2024-01-18:
The scrollbar appears in the Device, Project and Calibration tabs, not for "🏠", Prepare and Preview.

I attached screenshots of a different configuration than that I reported on initially:

Bambu Studio 1.8.4.51 on Windows 11 on a 4K monitor in portrait at 200% scale. In particular "3 Preview" the "Print" button is cut off even it there's 2160 horizontal pixels (!).  You can see which tabs have a scrollbar at the bottom of each screenshot.

# 1 🏠
[image]
# 2 Prepare
[image]
# 3 Preview
[image]
# 4 Device
[image]
# 5 Project
[image]
# 6 Calibrate
[image]

walterwongbbl, 2024-01-24:
@maruel 
Hello, I'm very sorry. Have you set your monitor to display vertically. We did not take this into consideration. We will solve this problem, but it will take some time.