bambulab/BambuStudio #3258: Text invisible in text tool while editing

Photogad, 2023-12-30:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 Pro

### Printer

X1C

### How to reproduce

1. Place text on the bottom of an object (bottom against build plate)
2. While putting the text in and editing it, it's invisible.
3. After exiting the text tool, the text finally shows up.

### Actual results

The text only is visible while NOT in the editor.

### Expected results

The text should be visible while editing it so you can see what the edits are doing.

This bug appeared in latest release -- previously in older versions of Bambu Studio, you could see the text while you are editing it.

This is very inconvenient.

### Project file & Debug log uploads

[fixme.zip](https://github.com/bambulab/BambuStudio/files/13799177/fixme.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Photogad, 2023-12-30:
The text should be visible while editing it so you can see what the edits are doing.

This bug appeared in latest release -- previously in older versions of Bambu Studio, you could see the text while you are editing it.

This is very inconvenient.

Haidiye01, 2024-01-02:
[image]
[image]
It is indeed a problem. We will fix it as soon as possible. At present, you can first access the bottom view through the interface or shortcut keys to view it

Haidiye01, 2024-01-09:
The problem has been resolved and  you can use it  in version 1.9.thank you~

closed 2024-05-06 by Haidiye00

Haidiye01, 2024-05-06:
ok,i have received your letter ,thank you!