bambulab/BambuStudio #3260: More temperature increments

Sonicboom247, 2023-12-31:
It's quite annoying trying to raise or lower the filament temperature from the touch screen during a print.  Since the only options are +- 5 or 30 degrees. I have to wait for it to go past a certain temperature, calculate the difference, subtract 5, wait for it to drop 4, add 5 just so it can go up one degree.

I would like to see either +- 1, +- 10 or just an option to input the desired temp for both the bed or extruder

as stated above, sometimes i will print the same file with a different filament and instead of reslicing, i change the temp on the display. I can.

**Additional context**
Add any other context or screenshots about the feature request here.

masc2279, 2023-12-31:
This is not an issue. You can use the handy app or the Bambu Studio and set what ever number you like in any incrament. And as well as the controller on most devices.

[image]

[image]

[image]

Sonicboom247, 2023-12-31:
This is in fact an issue on the A1

masc2279, 2023-12-31:
> This is in fact an issue on the A1

You did not fill out the report completly you did not mention anyting about version of device or software or anything that can help, other than a blanket ststement that clearly is wrong since it does work for some. Editing it after the fact and still not filling out the report completly does not fully help