bambulab/BambuStudio #3261: Ability to view/delete from SD card

Sonicboom247, 2023-12-31:
currently, the only way to view timelapse videos or photos from the SD card is to remove it and insert it into a PC. This is quite silly that the file structure is only one way.  I can constantly send files to the printer until the memory card is full. Then I need to remove the memory card to delete files manually. This is the polar opposite of what I wanted Bambu Studio and Bambu handy for.  It is simpler for me to go to my other brand of printer, and remove the thumb drive which does not need an adapter, delete or overwrite the old files, insert it into the printer and print. 

**Describe the solution you'd like**
I would like to either view and or delete files, and videos from the Bambu Handy or Bambu Studio.  either will work. or both.

**Describe alternatives you've considered**
Simply utilizing my old printer

**Additional context**
Add any other context or screenshots about the feature request here.

masc2279, 2023-12-31:
This is not an issue. It right in front in the handy app and Bambu Studio.  

You can use any platform to view files, videos and images that are on the printers memroy card and delete as well as download not only on mobile but on all platforms using Bambu Studio. 

Its a basic feature and it is not hidden in any way in either the Bambu Studio or the Handy App.

[image]

[image]

djengis, 2023-12-31:
Wait, that is possible? I can do this with my P1P and my A1?

Always thought this is X1C only.

I get: Initialize failed (Not supported on the current printer version)!
Booth have the latest firmware. Booth the same message.

[image]

Sonicboom247, 2023-12-31:
Thanks @djengis, that is exactly what i see.

masc2279, 2023-12-31:
x

> Thanks @djengis, that is exactly what i see, hence the reason for the request

You did not fill out the report completly you did not mention anyting about version of device or software or anything that can help, other than a blanket ststement that clearly is wrong since it does work for some.

Sonicboom247, 2024-01-01:
@masc2279, Nowhere on the request does it ask, mention, or indicate a device. My statement does indicate BOTH the Studio and the Handy app, Therefore, my statement is correct and your statement is in fact wrong since it does not work for some. You seem upset that your overuse of screenshots did not help.

masc2279, 2024-01-01:
@Sonicboom247  Tell me what is this ?

[image]

masc2279, 2024-01-01:
@Sonicboom247   I am not mad there should be more information thats all. Your blanket statement _"currently, the only way to view timelapse videos or photos from the SD card is to remove it and insert it into a PC."_ just is not true it might be on specific devices but its not on all. I would have just passed thru if it clearly stated what device 

Same as 
_"Describe the solution you'd like
I would like to either view and or delete files, and videos from the Bambu Handy or Bambu Studio. either will work. or both."_

Well it does work for some devices.

Just saying add more informations to the write up to be fully clear what you are using and where the ask is.

Sonicboom247, 2024-01-08:
Thanks for commenting but your link is to an old version that is not yet released for a different model printer. How does that make any sense? your link is for v1.8.0.57 with a P1, yet I am on 1.8.2.56 on an A1 with the latest firmware I can download and this option is still not there. so please, if anyone needs to do a quick search it's you.

Sonicboom247, 2024-01-09:
> You’re mixing studio version and firmware version. Studio has it baked in
> already, as of that previous studio release. What it’s waiting on is a new
> firmware version, which as described in those release notes is “to be
> released”. Sorry this confused you so terribly. You’ll just need to wait
> but it’s coming.

Nope, sorry. Not confused in the least as there is nothing confusing about
firmware and software versions. Good try though. The sad part is that some
people are even toxic on here, perhaps instead of feeble attempts to attack
someone posting for help or to improve a product, only the pertinent
information is posted and save your emotions for your therapist.

Sonicboom247, 2024-01-09:
> people are even toxic on here
>
> *as they are being toxic*... comical. Yeah, telling you to not post
> duplicate requests and to read the information that's available so you're
> not wasting anyone's time that could be better served resolving issues is
> not an attack but sure, play the victim. I'm done with you, keep being
> miserable.

Kkthxbai..lol

wsnjie, 2024-01-10:
This is still an issue as of today with the A1 on firmware 1.01.01.00 and Bambu Studio 1.8.3.89

[image]
[image]

Sonicboom247, 2024-01-28:
The latest firmware from today has fixed this issue. I can now read and delete from the SD card