bambulab/BambuStudio #3263: Bambu Studio Closes Abruptly When Trying to Run Manual Calibraiton

TechSavage1989, 2023-12-31:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Intel i9-11900k
GPU: RTX 4090
64GB Corsair Memory

### Printer

Bambu Lab A1

### How to reproduce

1. Go to "Calibration" Tab
[image]
2. Click "Manual Calibration"
[image]
- At this point, in the "Filament for Calibration " section, it just says "Incompatible in every slot and none are selectable. 
[image]
3. Nozzle Diameter has to be changed to something other than the default, then back again to have the Filament selection available and correct.
[image]
[image]
4. Clicking "Calibrate" causes the program to just shut down. 
[image]

### Actual results

Program just abruptly closes

This happens with any file or project with Both of my A1 Printers

### Expected results

I expect to be able to proceed with the calibration

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13800642/log.zip)
[mermaid-support-free-remix-robin3dverse.zip](https://github.com/bambulab/BambuStudio/files/13800643/mermaid-support-free-remix-robin3dverse.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

vinz486, 2024-01-02:
Same here: Appimage 1.8.2 on Ubuntu.

TechSavage1989, 2024-01-02:
I would like to add that I've now tried this across 4 separate machines running Windows 10 and 11 with identical results.

LiZ-Li-BBL, 2024-01-03:
We have fixed the issue where the filaments were incorrectly displayed as "Incompatible" and we are checking for crash issues. Please stay tuned for the next release.

TechSavage1989, 2024-02-02:
This issue is not resolved for me, and still occurs with he
latest firmware.