bambulab/BambuStudio #3265: Bambu Lab Studio missing P1S profiles

RedJohn14, 2023-12-31:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

macOS

### Printer

Bambu Lab P1S with AMS

### How to reproduce

Start Bambu Studio and select P1S Printer

### Actual results

Hello everyone,

I have noticed that there are no printer profiles for the P1S in the latest Bambu Studio version. When I select my printer (whether 0.2, 0.4 or other nozzle) I am only offered printer profiles for the X1C. I know that’s not bad in principle, but all other printers show the profiles with the correct name. Is this known? Will there be a solution (bugfix?)? Or is there already a beta where the problem is fixed, if so where exactly is the current beta?

Thank you very much!

### Expected results

All other printers have correct profiles except the P1S. It would be nice if this could be fixed. Or is there already a beta with a fix? With the otherwise so good features of Bambu Lab one can expect that even a 1000 $ printer with AMS is displayed correctly or? :-)

And read information from here that X1C profiles don't have the bug fix for the print bed:
https://forum.bambulab.com/t/bambu-lab-p1s-sd-card-access/44925

_These X1C profiles also don’t seem to have the buck fix for the print bed. When the print bed is far down at high prints, it moves to the bottom stop of the printer with a loud buzz noise. I continue to use my P1P profiles with the fan setting in the filament._

### Project file & Debug log uploads

Start Bambu Studio and select P1S Printer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

miru8112, 2024-01-02:
Same, this was very irritating to me when I first used it, encountered a problem and went on the hunt for it. Took me a while to understand that THIS wasn't the problem... Anyway, please give us our own specialised profiles...

XunZhangBambu, 2024-01-03:
Oh that's a feature . We share the profiles between p1s and x1c since they share the printing settings and this structure helps us construct the profiles better.The hardware differences will be reflected in the machine profiles.

miru8112, 2024-01-03:
> Oh that's a feature . We share the profiles between p1s and x1c since they share the printing settings and this structure helps us construct the profiles better.The hardware differences will be reflected in the machine profiles.

Interesting. Ok. Thank you for letting me know.

RedJohn14, 2024-01-03:
> Oh that's a feature . We share the profiles between p1s and x1c since they share the printing settings and this structure helps us construct the profiles better.The hardware differences will be reflected in the machine profiles.

@XunZhangBambu 

Thank you for the explanation! Are there plans to provide a separate profile for the P1S anyway? Basically, of course, it is similar to the X1C. But it would still be nice to have a separate profile for such an expensive printer. 

Especially at the beginning when you start the slicer for the first time it is very confusing.

The P1P also has its own profile, even if it "only" lacks the housing and a few built-in parts, it still has its own profile.

I and I think many others would be very happy if a separate profile could be realized.

The A1 and A1 mini also have their own profiles, even though they are "identical" printers with a larger print area :-)

As you can see, I would be very happy if the whole thing could be realized. In terms of programming, it's probably not too much effort, I would estimate as a "non-programmer" ^^.

XunZhangBambu, 2024-01-04:
@RedJohn14 
Hello, in terms of workload, adding a separate profile for an individual printer isn't too challenging. However, it might reduce the reusability of profiles due to the presence of many duplicated parameters. The problem is splitting the previously shared presets might cause issues with users opening their previously saved 3MF files. This is because even if the preset name is modified, the 3MF file still stores the old preset name. Sharing presets doesn't impact print quality. In fact, we've already confirmed that it ensures a high print quality when deciding whether to share presets between printers. When sharing isn't feasible, we create individual presets, such as the recently released Bambu TPU 95A HF.

CopaceticMeatbag, 2024-04-02:
I agree it makes much more sense not creating a separate profile when the settings will be identical - perhaps we could slightly adjust the name scheme to something like `0.3mm Standard @BBL X1C/P1S 0.6 nozzle` ?

closed 2024-06-26 by XunZhangBambu

Fixitdaz, 2025-05-18:
Yes at least add the name of supported printer to the profile like @bbl /x1c/p1s/p1s

CopaceticMeatbag, 2025-11-04:
>  splitting the previously shared presets might cause issues with users opening their previously saved 3MF files.

This will be such a transient issue. And using the same logic, **not** splitting the previously shared presets **does** cause issues with users not seeing a matching profile for their printer.

ERisEPR, 2026-10-22:
Can someone change the preset name from "X1C" to "P1S/X1C"