bambulab/BambuStudio #3272: LAN mode does not work between VLANs

EstrellaXD, 2024-01-01:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.2.1

### Additional system information

CPU: M3Pro
Mem: 18GB

### Printer

Bambu A1/A1 mini

### How to reproduce

In the same local area network, the host and the printer are in different VLANs, and the host cannot find and connect to the printer.

### Actual results

Same as above.

### Expected results

I found that using `espressif.` broadcasting instead of using `.local` mDNS broadcasting may hinder the process of mDNS mirroring forwarding by the upper-level gateway.

### Project file & Debug log uploads

In same VLAN
```
ping espressif.
PING espressif (192.168.1.12): 56 data bytes
64 bytes from 192.168.1.12: icmp_seq=0 ttl=255 time=18.257 ms
64 bytes from 192.168.1.12: icmp_seq=1 ttl=255 time=22.654 ms
64 bytes from 192.168.1.12: icmp_seq=2 ttl=255 time=28.148 ms
64 bytes from 192.168.1.12: icmp_seq=3 ttl=255 time=68.280 ms
```

Other VLAN

```
ping espressif.
ping: cannot resolve espressif.: Unknown host
```

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-01-02:
Hi. @EstrellaXD 
we do not yet support mDNS, But we will consider supporting it

wilk-polarny, 2024-01-04:
It's not necessary different VLANs. The printer will only advertise within the same subnet (which makes perfect sense, since routers will not let broadcasts traverse).
So if there's for example a net with PCs running bambu studio on 192.168.1.0/24 (with a gateway at the same address) and the printer in a different subnet for IoT devices - let's say 192.168.2.0/24 - the printer will be only visible if Bambu Studio is being run within the 192.168.2.0/24 subnet.

It would really help to have a **MANUAL IP entry option** in Bambu Studio - the option of adding LAN printers by IP and save them. This would help enormously as a fallback if auto detection does not work (and you know your printer's IP addresses or even have reserved addresses).

EstrellaXD, 2024-01-04:
> It's not necessary different VLANs. The printer will only advertise within the same subnet (which makes perfect sense, since routers will not let broadcasts traverse). So if there's for example a [...]

Routers or switches that support three-layer forwarding usually support mDNS mirroring. Alternatively, you can manually set firewall rules to allow mDNS to penetrate different VLANs. I don't think this is a problem.

However, in enterprise or advanced home networks, it is very common to place production hosts and IoT devices such as 3D printers in different VLANs. This helps protect the LAN network security and allows for targeted firewall rule settings.

> It would really help to have a MANUAL IP entry option in Bambu Studio - the option of adding LAN printers by IP and save them. This would help enormously as a fallback if auto detection does not work (and you know your printer's IP addresses or even have reserved addresses).

I agree with your statement that manually setting the IP can solve this problem to a certain extent, but I believe that using mDNS would be more elegant and reliable.

cmmh, 2024-10-25:
Unfortunately, creating a bunch of mDNS mirrors/proxies isn't the right answer here.  The right answer is for Bambu slicer's networking plugin to accept an ip address.