bambulab/BambuStudio #3275: Heat bed changes temperature to match support interface layer

camfarris, 2024-01-01:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Slice print with supports where base and interface layer are 2 different materials (example: PETG base and PLA interface layer)
2. Begin print
3. Wait for print to start an interface layer

### Actual results

When swapping to interface layer, both the nozzle and heat bed change to the interface layer material's temperature instead of the base temperature

### Expected results

Only the nozzle temperature changes to reflect the interface layer material's temperature. The heat bed remains the base material temperature.

### Project file & Debug log uploads

Behavioral expectation issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2024-01-03:
Hello,you mean do not change heat bed temperature while switching to print interface layer with a different filament ?

camfarris, 2024-01-03:
@XunZhangBambu 

Yes. The heat bed should not change when swapping to the interface layer. This is because the interface layer does not touch the heat bed. The temperature should match the base material temperature.

Schokobecher, 2024-01-11:
Current work around is to make a material profile with the same bed temperature. But I agree, it should be at least a toggle/dropdown for the behavior.

scobby, 2024-05-29:
Hey there, having the exact same behavior. Only solutions for me was to set the pla filament bed temperature temporary for the print to the temperature of the PETG.
Based on the youtube videos around the topic how to get perfect supports with PLA/PETG i thing this is a commen problem for many people why prints fail in this combination. And its hard to track the problem to the cause.
Could you please fix this in the next release? @XunZhangBambu

david-breidert, 2025-04-04:
Commenting to bump this issue. This is still happening and with the release of the H2D even more important to fix.

mean-ui-thread, 2025-05-21:
I don't know for certain until I experiment a bit more, but every time I use Bambu support material with PETG filament, my layers get misaligned on the same layers than the support/raft interface layers. If I disable support, then my PETG prints are fine (except for the overhangs obviously)

My custom PETG filament setting sets the bed temperature to 77C, and Bambu support filament drops it to 60C, which is most likely the root cause of my print issue I believe.

[image]

[image]

[image]

For now, I will manually tweak the bed temperature of Bambu support filament to ensure that the bed temperature doesn't drop, and will do some tests to see if that solves the issue. If the resolves the issue, than yes, I agree 100%. the support filament should not modify the bed temperature. This should be automatic.

Bambu is advertising their support filament for both PLA and PETG but its settings are only for PLA prints. They really dropped the ball on that one.