bambulab/BambuStudio #3277: Avoid floating infil

kfsone, 2024-01-02:
Printing an object with an overhanging lip can result in the slicer generating g-code which tries to place material in unsupported space, which you can offset by turning on print infill first, but this may still try to print the unsupported filament *first*

[image]

[image]

In this scenario, it would clearly be better if gcode had first laid down the infill that had material underneath it. it should be possible to partition the infill based on this rule:

- has material underneath > N% of line or is end of supported line/loop that extends < %M mm from material (e.g rectilinear lines that are part of a slope)

Lay down this material first, then for the remaining material, prioritize it by proximity to already-printed material on the same layer, this would cause the outer edge of this lip to be printed outward infill by infill line.

Project attached:
[Rectangular Water Bottle Lid - P1S.zip](https://github.com/bambulab/BambuStudio/files/13806424/Rectangular.Water.Bottle.Lid.-.P1S.zip)

RaphTronic, 2024-01-04:
+1 on this request that results in print FAILS. Although I would classify it as a BUG, not a feature.

Has happened on several of my models as well. The slicer sometimes lays down infill lines in the air instead of starting from the closest wall it is going to connect to. The order should simply be reversed. Makes no sense and makes several prints fail.

hotellonely, 2025-01-31:
[image]

terrible :/