bambulab/BambuStudio #3279: Liveview broken when building from source

hadess, 2024-01-02:
### Bambu Studio Version

git

### Where is the application from?

Built from source by yourself or others

### OS version

Linux

### Printer

N/A

### How to reproduce

Build BambuStudio with the gst-plugins-bad headers installed

### Actual results

Liveview not working

### Expected results

Liveview working

### Project file & Debug log uploads

The problem is that the liveview code (in `src/slic3r/GUI/wxMediaCtrl2.cpp`) in BambuStudio expects to be able to access the `m_playbin` object member which is only available when wxWidgets is built with GStreamer support but **not GstPlayer support**.

You should be able to reproduce the problem with AppImage by installing `libgstreamer-plugins-bad1.0-dev` in the container, then wxWidgets will be built with the GstPlayer backend (in `src/unix/mediactrl_gstplayer.cpp`) rather than the playbin-based backend (in `src/unix/mediactrl.cpp`). Only the latter works with BambuStudio.

The liveview code in BambuStudio should be adapted to work with both backends.

In the meantime, the Flatpak build will forcefully disable the GstPlayer-backend by removed `set(wxUSE_GSTREAMER_PLAYER ON)` from `build/cmake/init.cmake` in wxWidgets.

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-01-03:
Okay removed set (wxUSE_GSTREAMER_PLAYER ON). I think that's the only thing we can do in the near future.

hadess, 2024-08-06:
Poke